immortalwrt/immortalwrt #1730: IPv6 does not work on Ariaboard Photonicat

neobar, 2025-02-21:
### Describe the bug

I use the 5G modem for network connection. Everything is fine with 23.05.4, but IPv6 does not work on 24.10.0.

The clients can get an ipv6 address but no ipv6 connection. The ipv6 connection is ok on the router.

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Ariaboard Photonicat

### Image kind

Official downloaded image

### Steps to reproduce

Install luci-proto-mbim
```
opkg install luci-proto-mbim
```

Add following lines to `/etc/config/network`
```
config interface 'lte'
	option proto 'mbim'
	option device '/dev/cdc-wdm0'
	option apn 'CTNET'
	option auth 'none'
	option pdptype 'ipv4v6'
```

Configure IPv6 of lan
```
root@ImmortalWrt:~# cat /etc/config/dhcp

config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option min_cache_ttl '3600'
	option use_stale_cache '3600'
	option cachesize '8000'
	option nonegcache '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option dns_redirect '1'
	option ednspacket_max '1232'
	option filter_aaaa '0'
	option filter_a '0'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option ra 'server'
	option max_preferred_lifetime '2700'
	option max_valid_lifetime '5400'
	list ra_flags 'none'
	option dns_service '0'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

### Actual behaviour

From router
```
ping test6.ustc.edu.cn
PING test6.ustc.edu.cn (2001:da8:d800::1043): 56 data bytes
64 bytes from 2001:da8:d800::1043: seq=0 ttl=50 time=65.207 ms
64 bytes from 2001:da8:d800::1043: seq=1 ttl=50 time=85.191 ms
^C
--- test6.ustc.edu.cn ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 65.207/75.199/85.191 ms
```

From clients
```
ping6 test6.ustc.edu.cn
PING6(56=40+8+8 bytes) 2409:8d30:4a:4dbe:fd61:5a9c:f2b4:179d --> 2001:da8:d800::1043
^C
--- test6.ustc.edu.cn ping6 statistics ---
5 packets transmitted, 0 packets received, 100.0% packet loss
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

initzhu, 2025-02-21:
Solution: Select your Interfaces in Network>Interfaces, click "Edit", find Extend prefix in General Settings, select and apply/

[image]

neobar, 2025-02-21:
> Solution: Select your Interfaces in Network>Interfaces, click "Edit", find Extend prefix in General Settings, select and apply/
> 
> [image]

I do not thinks this is the right solution. With 5G modem, the `wan` and `wan6` interfaces are unused. The virtual dynamic interfaces `lte_4` and `lte_6` are not editable. The `lte` interface itself does not have the `Extend prefix` option.

[image]

neobar, 2025-02-23:
I have found the caus of this issue. There is an incorrect ipv6 route.
```
root@ImmortalWrt:~# ip -6 r
2409:8d30:4c:f63d::/64 dev wwan0  metric 256
2409:8d30:4c:f63d::/64 dev br-lan  metric 1024
```

The command `ip -6 route del 2409:8d30:4c:f63d::/64 dev wwan0` can temporarily solve the problem.

The question still remains as to how such a wrong route was inserted into the routing table.

neobar, 2025-03-03:
A simpler solution.
```bash
uci set network.lan.metric='127'
uci commit network
service network restart
```

gaifeng8864, 2025-03-03:
My PC is working fine, but the Android device has an IPv6 address, but the IPv6 cannot connect to the Internet. It can occasionally return to normal after restarting or reconnecting to WiFi. This problem is common on openwrt versions that do not enable NSS acceleration. I have tested that there is no such problem on openwrt with NSS acceleration. I suspect it may be related to the compatibility of the wireless driver.

rbqvq, 2025-03-12:
Try use `luci-proto-modemmanager`

closed 2025-04-22 by neobar