immortalwrt/immortalwrt #1732: tr3000 v1 24.10.0 uboot sysupgrade镜像无法刷入

huangzhiguo, 2025-02-22:
### Describe the bug

按照
https://github.com/immortalwrt/immortalwrt/commit/51272fcd004955198265f030707aac1dd9794b98
中的instruction，依次刷入了immortalwrt官网下载的24.10.0 release版的下面四个文件。
cudy_tr3000-v1-ubootmod-preloader.bin
cudy_tr3000-v1-ubootmod-bl31-uboot.fip
cudy_tr3000-v1-ubootmod-initramfs-recovery.itb
cudy_tr3000-v1-ubootmod-squashfs-sysupgrade.itb
前三个都正常刷入，但sysupgrade固件刷入后重启，依然是initramfs固件，无论使用web界面升级还是sysupgrade -n -F命令都是一样；
好在我备份了六个分区，恢复了出厂固件；
抱歉，没有日志可以提供，我的是中文包装的tr3000v1。另一个小插曲是恢复出厂固件不能选择官方谷歌网盘里的recovery.bin，需要用客服提供的固件才能成功。

### ImmortalWrt version

24.10.0

### ImmortalWrt release

not available

### ImmortalWrt target/subtarget

not available

### Device

cudy TR3000 v1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

initzhu, 2025-02-22:
可以先用原厂布局(非openwrt layout)的
[uboot可以刷这个](https://drive.wrt.moe/uboot/mediatek/mt7981-cudy_tr3000-v1-fip-fit.bin)
如果您能克服困难使用那个反人类串口，提供ttl日志 就更好了

huangzhiguo, 2025-02-22:
> 可以先用原厂布局(非openwrt layout)的 [uboot可以刷这个](https://drive.wrt.moe/uboot/mediatek/mt7981-cudy_tr3000-v1-fip-fit.bin) 如果您能克服困难使用那个反人类串口，提供ttl日志 就更好了

我不是硬件工程师，不会用TTL，而且需要拆机。之前刷坏了一个找客服换了。这个uboot不清楚细节，不敢乱刷，要不然变砖就浪费了。

1715173329, 2025-02-22:
修了。

closed 2025-02-22 by 1715173329

huangzhiguo, 2025-02-22:
> 修了。就一行，现有版本升级的时候自己改一下对应的文件就行了。

确实是这样，我加了这行，刷成功了

Gurosuogen, 2025-03-02:
> > 修了。就一行，现有版本升级的时候自己改一下对应的文件就行了。
> 
> 确实是这样，我加了这行，刷成功了

我也出现了同样的问题，固件是在 firmware-selector 中选的 24.10.0 版本的 [Cudy TR3000 v1 (OpenWrt U-Boot layout)] , 请问您是如何解决这个问题的？

huangzhiguo, 2025-03-03:
> > > 修了。就一行，现有版本升级的时候自己改一下对应的文件就行了。
> > 
> > 
> > 确实是这样，我加了这行，刷成功了
> 
> 我也出现了同样的问题，固件是在 firmware-selector 中选的 24.10.0 版本的 [Cudy TR3000 v1 (OpenWrt U-Boot layout)] , 请问您是如何解决这个问题的？

ssh登录路由器
vi /lib/upgrade/platform.sh
根据bug修复的内容修改，保存后即可升级。注意，保存后不能重启路由器，否则白耽误功夫

Gurosuogen, 2025-03-03:
> > > > 修了。就一行，现有版本升级的时候自己改一下对应的文件就行了。
> > > 
> > > 
> > > 确实是这样，我加了这行，刷成功了
> > 
> > 
> > 我也出现了同样的问题，固件是在 firmware-selector 中选的 24.10.0 版本的 [Cudy TR3000 v1 (OpenWrt U-Boot layout)] , 请问您是如何解决这个问题的？
> 
> ssh登录路由器 vi /lib/upgrade/platform.sh 根据bug修复的内容修改，保存后即可升级。注意，保存后不能重启路由器，否则白耽误功夫

感谢解答，修改之后已成功运行！