immortalwrt/immortalwrt #1733: PPTP client failed to start

hellobbq, 2025-02-22:
### Describe the bug

```log
daemon.notice netifd: Interface 'vpn_1' is setting up now
daemon.err modprobe: no module folders for kernel version 6.8.12-8-pve found #<-- This issue also exists on openwrt 23.05.3, but it does not affect normal activation.
daemon.info pppd[29561]: Plugin pptp.so loaded.
daemon.info pppd[29561]: PPTP plugin version 1.00
daemon.err pppd[29561]: In file /etc/ppp/options.pptp: unrecognized option 'mppe'
daemon.notice netifd: vpn_1 (29561): /usr/sbin/pppd: In file /etc/ppp/options.pptp: unrecognized option 'mppe'
daemon.notice netifd: Interface 'vpn_1' is now down
```

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

x86/64

### Device

Default string

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

tian4546, 2025-05-19:
这个能处理吗，用ai回复是缺失ppp-pptp-mppe 插件

1715173329, 2025-05-26:
Try to apply this commit: https://github.com/x-wrt/x-wrt/commit/0d50ea29a3b90c0049236fb1d20862bee6e95a98

commit cdeb7e41 (2025-05-27): ppp: update pppd options

Fixes: #1733
closed 2025-05-27 by 1715173329