immortalwrt/immortalwrt #1749: Cudy TR3000 多酷TR3000下发的DHCP IP是wan口IP，未和lan口处于同一网段

aaro-n, 2025-03-07:
### Describe the bug

使用中兴F50（192.168.10.1）通过8153B网卡用网线连接到TR3000的wan口，TR3000的网关IP为192.168.60.1，通过WI-FI或网线连接到TR3000时，所分配的IP为192.168.10.X，当断开F50网线连接，通过WI-FI或网线连接到TR3000时此时分配的IP为192.168.60.X，再次连接F50网线，此时可以正常上网（192.160.60.X）但如果重新连接到TR3000，此时又会分配IP为192.168.10.X，F50和TR3000都开启了DHCP，感觉TR3000的DHCP优先级没有F50高，友善的R3S就没这个问题。
已经关闭TR3000的DHCP IPV6，但手机或电脑依旧可以收到分配的公网IPv6，F50未关闭IPv6.

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1 (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Steps to reproduce

1，将F50通过8153B网卡转网线连接TR3000
2. 通过WI-FI或网线连接TR3000，此时分配的IP为192.168.10.X
3. 拔掉TR3000网线，再次连接TR3000，此时分配为192.168.60.X
4. 此时可以再次连接F50网线，可以正常上网（192.168.60.X），如果断开TR3000连接，重连TR3000，此时分配的IP又为192.168.60.X

### Actual behaviour

TR3000未分配192.168.60.X而是分配192.168.10.X

### Expected behaviour

TR3000为连接到路由器的设备分配lan口IP，即192.168.60.X

除了F50外也尝试连接光猫也是相同的问题

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

lylus, 2025-03-09:
@aaro-n 大佬，你有没有试过用usb连接？新版24.10用.itb格式的固件，启动感觉比以前23.05的慢，导致中兴F50开机时没法检测到usb口有连接设备，就不开起usb共享。必需每次启动路由器后要重启一下F50才能认到F50的网卡。目前头疼死了，只能用回以前23.05的老固件。

aaro-n, 2025-03-10:
@lylus 没有，我是通过网线连接F50的，直接用USB嫌不稳定。