immortalwrt/immortalwrt #1753: zte 8820s 刷入24.10和23.05.4都会无限重启，进不去

lovejoy, 2025-03-11:
### Describe the bug

breed 是https://breed.hackpascal.net/breed-mt7621-zte-e8820s.bin
initramfs的能进去，factory-squashfs的灯会亮几秒，然后就会无限重启

### ImmortalWrt version

24.10

### ImmortalWrt release

24.10

### ImmortalWrt target/subtarget

mt7621

### Device

zte 8820s

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-03-11:
85e0f369a080d38b2c4728c8048f10a6c8938cd9

closed 2025-03-11 by 1715173329

lovejoy, 2025-03-11:
了解了。要求必须用 https://firmware.download.immortalwrt.eu.org/uboot/ramips/mt7621-zte_e8820s-u-boot.bin  这个uboot。