immortalwrt/immortalwrt #1757: Orange pi 5 16G NVME

basic-dev-000, 2025-03-14:
### Describe the bug

Oranger pi 5 board version 1.3.2 16G RAM. Unable to boot via nvme using u-boot-rockchip-spi.bin.
Working on Oranger pi 5 board version v1.2 4G RAM

### ImmortalWrt version

Snapshot

### ImmortalWrt release

Snapshot

### ImmortalWrt target/subtarget

Snapshot

### Device

Snapshot

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-03-15:
try 60990ba523d8e3563e1df07f4db3f72b937b4e78

basic-dev-000, 2025-03-16:
> try [60990ba](https://github.com/immortalwrt/immortalwrt/commit/60990ba523d8e3563e1df07f4db3f72b937b4e78)

I tried, but no locked.

basic-dev-000, 2025-03-16:
Here's logs Orange pi 5 board lower version. spi nvme working.
U-Boot 2024.10-User-Soft-r32868-1ceaac207b (Jan 03 2025 - 12:22:24 +0000)

Model: Xunlong Orange Pi 5
DRAM:  4 GiB
Core:  328 devices, 32 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 0
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: Xunlong Oranga Pi 5
Net:   eth0: ethernet@fe1c0000
Hit any key to stop autoboot:  0
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
Card did not respond to voltage select! : -110
No EFI system partition
No EFI system partition
Failed to persist EFI variables
Scanning bootdev 'mmc@fe2c0000.bootdev':
Card did not re3pond to voltage select! : -110
No EFI sy3tem partition
Failed to persist EFI variables
No EFI system partition
Failed to persist EFI variables

Scannin' bootdev 'nvme#0.blk#1.bootdev':
  0  script       ready   nvme         1  nvme#0.blk#0.bootdev.part /boot.scr
** Booting bootflow 'nvme#0.blk#1.bootdev.part_1' with script
4841696 bytes read in 14 ms (329.8 MiB/s)
## Loading kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.6.69
     Type:         Kernel Image
     Compression:  lzma compr%ssed
                                       Data Start:   0x020000e8
     Data Size:    4765178 Bytes = 4.5 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x03200000
     Entry Point:  0x03200000
     Hash algo:    crc32
     Hash value:   df4fa14d
     Hash algo:    sha1
     Hash value:   ac4$01884510d477e75e7faa42d5671710705cd4
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'fdt-1' fdt subimage
     Description:  ARM64 OpenWrt xunlong_orangepi-5 device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x0248b828
     Data Size:    74607 Bytes = 72.9 KiB
     Architecture: AArch64
     Hash algo:    crc32
     Hash value:   82938aab
     Hash algo:    sha1
     Hash value:   5c843c326237be8ce428bff9b58b7798ffdf5d55
   Verifyinga3h Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x248b828
Working FDT set to 248b828
   Uncompressing Kernel Image to 3200000
   Loading Device Tree to 00000000ece92000, end 00000000ecea736e ... OK
Working FDT set to ece92000

Starting kernel ...

basic-dev-000, 2025-03-16:
Here's newer  board v1.3.2 logs orange pi 5 not worked.

U-Boot 2024.10-User-r32868-1ceaac207b (Jan 03 2025 - 12:22:24 +0000)

Model: Xunlong Orange Pi 5
DRAM:  16 GiB
Core:  328 devices, 32 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 0
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: Xunlong Orange Pi 5
Net:   eth0: ethernet@fe1c0000
Hit any key to stop autoboot:  0
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ------------                                                                                                                                   ----
Scanning global bootmeth 'efi_mgr':
Card did not respond to voltage select! : -110
No EFI system partition
No EFI system partition
Failed to persist EFI variables
Scanning bootdev 'mmc@fe2c0000.bootdev':
Card did not respond to voltage select! : -110

pcie_dw_rockchip pcie@fe190000: PCIe-0 Link Fail
pcie_dw_rockchip pcie@fe190000: PCIe-0 Link Fail

scanning bus for devices...
Bus usb@fc800000: USB EHCI 1.00
Bus usb@fc840000: USB OHCI 1.0
Bus usb@fc880000: USB EHCI 1.00
Bus usb@fc8c0000: USB OHCI 1.0
Bus usb@fcd00000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fc800000 for devices... 1 USB Device(s) found
scanning bus usb@fc840000 for devices... 1 USB Device(s) found
scanning bus usb@fc880000 for devices... cannot reset port 1!?
1 USB Device(s) found
scanning bus usb@fc8c0000 for devices... 1 USB Device(s) found
scanning bus usb@fcd00000 for devices... 1 USB Device(s) found
pcie_dw_rockchip pcie@fe190000: PCIe-0 Link Fail
ethernet@fe1c0000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
phy_startup() failed: -110
FAILED: -110
pcie_dw_rockchip pcie@fe190000: PCIe-0 Link Fail
ethernet@fe1c0000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
phy_startup() failed: -110
FAILED: -110
No more bootdevs
---  -----------  ------  --------  ----  ------------------------  ----------------
(0 bootflows, 0 valid)

basic-dev-000, 2025-03-16:
I think the NVMe PCI driver did not work.

1715173329, 2025-03-16:
I don't know where you get the uboot, but it's not the one from ImmortalWrt, so you are on your own.

basic-dev-000, 2025-03-16:
> I don't know where you get the uboot, but it's not the one from ImmortalWrt, so you are on your own.

From here: immortalwrt/build_dir/target-aarch64_generic_musl/u-boot-orangepi-5-rk3588s/u-boot-2024.10/u-boot-rockchip-spi.bin
and write to opi 5: dd if=u-boot-rockchip-spi.bin of=/dev/mtdblock0 conv=noerror,sync

1715173329, 2025-03-16:
snapshots uses uboot 2025.01, not 2024.10

basic-dev-000, 2025-03-16:
> snapshots uses uboot 2025.01, not 2024.10

same error: [image]

1715173329, 2025-03-16:
Please report this issue to [u-boot mailing list](https://lore.kernel.org/u-boot): `u-boot@lists.denx.de` and Cc `Jonas Karlman <jonas@kwiboo.se>`.

basic-dev-000, 2025-03-16:
> Please report this issue to [u-boot mailing list](https://lore.kernel.org/u-boot): `u-boot@lists.denx.de` and Cc `Jonas Karlman <jonas@kwiboo.se>`.

Okay, thankyou