immortalwrt/immortalwrt #1759: 24.10 日志中经常出现br-lan重启

beck023, 2025-03-15:
### Describe the bug

root@x86_Wrt:~# cat /etc/openwrt_release
DISTRIB_ID='ImmortalWrt'
DISTRIB_RELEASE='24.10.0'
DISTRIB_REVISION='r32824-6a73dae98c9c'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='ImmortalWrt 24.10.0 r32824-6a73dae98c9c'
DISTRIB_TAINTS=''

**ESXi-7.0U3q 直通intel i350-T4的两个网口,LAN/WAN各一个
全新安装24.10后日志中经常出现lan口掉线重启, 之前一直从21.02.0到23.05.4 没出现过同样的问题.**

Sat Mar 15 19:45:26 2025 daemon.notice netifd: Network device 'eth0' link is down
Sat Mar 15 19:45:26 2025 kern.info kernel: [ 1153.048780] igb 0000:0b:00.0 eth0: igb: eth0 NIC Link is Down
Sat Mar 15 19:45:26 2025 kern.info kernel: [ 1153.049142] br-lan: port 1(eth0) entered disabled state
Sat Mar 15 19:45:27 2025 daemon.notice netifd: bridge 'br-lan' link is down
Sat Mar 15 19:45:27 2025 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sat Mar 15 19:45:27 2025 daemon.warn odhcpd[2378]: No default route present, setting ra_lifetime to 0!
Sat Mar 15 19:45:30 2025 daemon.notice netifd: Network device 'eth0' link is up
Sat Mar 15 19:45:30 2025 daemon.notice netifd: bridge 'br-lan' link is up
Sat Mar 15 19:45:30 2025 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Mar 15 19:45:30 2025 kern.info kernel: [ 1156.799968] igb 0000:0b:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sat Mar 15 19:45:30 2025 kern.info kernel: [ 1156.802126] br-lan: port 1(eth0) entered blocking state
Sat Mar 15 19:45:30 2025 kern.info kernel: [ 1156.802394] br-lan: port 1(eth0) entered forwarding state

Sat Mar 15 21:49:16 2025 daemon.notice netifd: Network device 'eth0' link is down
Sat Mar 15 21:49:16 2025 kern.info kernel: [ 8582.859713] igb 0000:0b:00.0 eth0: igb: eth0 NIC Link is Down
Sat Mar 15 21:49:16 2025 kern.info kernel: [ 8582.860087] br-lan: port 1(eth0) entered disabled state
Sat Mar 15 21:49:17 2025 daemon.notice netifd: bridge 'br-lan' link is down
Sat Mar 15 21:49:17 2025 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sat Mar 15 21:49:17 2025 daemon.warn odhcpd[2378]: No default route present, setting ra_lifetime to 0!
Sat Mar 15 21:49:20 2025 daemon.notice netifd: Network device 'eth0' link is up
Sat Mar 15 21:49:20 2025 daemon.notice netifd: bridge 'br-lan' link is up
Sat Mar 15 21:49:20 2025 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Mar 15 21:49:20 2025 kern.info kernel: [ 8587.000933] igb 0000:0b:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sat Mar 15 21:49:20 2025 kern.info kernel: [ 8587.003143] br-lan: port 1(eth0) entered blocking state
Sat Mar 15 21:49:20 2025 kern.info kernel: [ 8587.003418] br-lan: port 1(eth0) entered forwarding state

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware7,1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

loveritsu929re, 2025-05-05:
Same (link down, interface disabled, happened to all br-lan, wan, wan_6); Network is down every morning and I have to restart immortalwrt.
I noticed ttyd was complaining in the syslog, so I `apk del ttyd` (I don't have the luci app). Now it finally starts to work overnight.

dutian312, 2025-11-03:
以前我遇到过有个设置会导致这个问题，不知道是否安装了或者使用了内存释放工具/命令，有的话去掉。