immortalwrt/immortalwrt #1763: 24.10 Restore backup "The uploaded backup archive is not readable"

hanakoNya, 2025-03-18:
### Describe the bug

**Bug Report: Backup Restore Functionality Issue**

When using the backup restore feature in version 24.10, I encountered the message "The uploaded backup archive is not readable" after uploading the "backup.tar" backup file. Testing revealed that even when using a "backup.tar" file generated in version 24.10, the same message appears. However, I can successfully extract the contents using the command "tar xvf backup.tar" in the SSH console. This issue does not occur in version 23.05.

### ImmortalWrt version

r32963-75a1bff338

### ImmortalWrt release

24.10-SNAPSHOT

### ImmortalWrt target/subtarget

mediatek/filogic

### Device

H3C Magic NX30 Pro (NMBM layout)

### Image kind

Official downloaded image

### Steps to reproduce

1. Backup to the "backup.tar" file in version 24.10 and download it to the local machine.
2. Select the "backup.tar" file locally and upload it.
3. The message "The uploaded backup archive is not readable" is reproduced.

### Additional info

Upload  backup Use macOS Safari.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-03-18:
#1622

skefy, 2025-03-19:
https://www.reddit.com/r/openwrt/comments/rbdzl9/openwrt_backups_dont_work_the_uploaded_backup/

好像是Safari浏览器下载之前解压缩了 tar.gz 文件，变成了一个tar压缩包。 需要自己手动压缩成tar.gz就行了