immortalwrt/immortalwrt #1764: Cudy TR3000刷immortalwrt 24.10后USB连接F50随身WI-FI接口持续断开（设备未自动重启）

h2ch, 2025-03-19:
### Describe the bug

固件版本：ImmortalWrt 24.10.0 r32824-6a73dae98c9c (OpenWrt U-Boot layout)
软件包：kmod-usb-core kmod-usb-ehci kmod-usb-net kmod-usb-net-cdc-ether kmod-usb-net-rndis kmod-usb-xhci-mtk kmod-usb3
系统日志：
Wed Mar 19 02:01:15 2025 daemon.notice netifd: Interface 'USBWAN' is now up
Wed Mar 19 02:01:16 2025 kern.info kernel: [  846.759137] usb 2-1: USB disconnect, device number 22
Wed Mar 19 02:01:16 2025 kern.info kernel: [  846.759371] rndis_host 2-1:1.0 eth2: unregister 'rndis_host' usb-11200000.usb-1, ZTE RNDIS device
Wed Mar 19 02:01:16 2025 kern.warn kernel: [  846.760066] xhci-mtk 11200000.usb: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Wed Mar 19 02:01:16 2025 kern.info kernel: [  847.159795] usb 2-1: new SuperSpeed USB device number 23 using xhci-mtk
Wed Mar 19 02:01:16 2025 kern.info kernel: [  847.207087] rndis_host 2-1:1.0 eth2: register 'rndis_host' at usb-11200000.usb-1, ZTE RNDIS device, 10:3c:59:de:f1:f1
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Interface 'USBWAN' is disabled
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Network device 'eth2' link is down
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Interface 'USBWAN' has link connectivity loss
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9343): udhcpc: received SIGTERM
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9343): udhcpc: unicasting a release of 172.16.0.101 to 172.16.0.1
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9343): udhcpc: sending release
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9343): udhcpc: bind(UDP): Address not available
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9343): udhcpc: entering released state
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9343): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "USBWAN" } (Permission denied)
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Interface 'USBWAN' is now down
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Interface 'USBWAN' is enabled
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Network device 'eth2' link is up
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Interface 'USBWAN' has link connectivity
Wed Mar 19 02:01:16 2025 daemon.notice netifd: Interface 'USBWAN' is setting up now
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9633): udhcpc: started, v1.36.1
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9633): udhcpc: broadcasting discover
Wed Mar 19 02:01:16 2025 daemon.notice netifd: USBWAN (9633): udhcpc: broadcasting select for 172.16.0.101, server 172.16.0.1
Wed Mar 19 02:01:17 2025 daemon.notice netifd: USBWAN (9633): udhcpc: lease of 172.16.0.101 obtained from 172.16.0.1, lease time 43199
Wed Mar 19 02:01:17 2025 daemon.notice netifd: Interface 'USBWAN' is now up
Wed Mar 19 02:01:17 2025 daemon.notice netifd: Network device 'eth2' link is down
Wed Mar 19 02:01:17 2025 daemon.notice netifd: Interface 'USBWAN' has link connectivity loss
Wed Mar 19 02:01:17 2025 kern.info kernel: [  848.258775] usb 2-1: USB disconnect, device number 23
Wed Mar 19 02:01:17 2025 kern.info kernel: [  848.258945] rndis_host 2-1:1.0 eth2: unregister 'rndis_host' usb-11200000.usb-1, ZTE RNDIS device
Wed Mar 19 02:01:17 2025 kern.warn kernel: [  848.259127] xhci-mtk 11200000.usb: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Wed Mar 19 02:01:18 2025 kern.info kernel: [  848.678779] usb 2-1: new SuperSpeed USB device number 24 using xhci-mtk
Wed Mar 19 02:01:18 2025 kern.info kernel: [  848.715039] rndis_host 2-1:1.0 eth2: register 'rndis_host' at usb-11200000.usb-1, ZTE RNDIS device, 10:3c:59:de:f1:f1
Wed Mar 19 02:01:18 2025 daemon.notice netifd: Interface 'USBWAN' is disabled

内核日志：
[  710.849649] usb 1-1: new full-speed USB device number 2 using xhci-mtk
[  711.029878] usb 1-1: not running at top speed; connect to a high speed hub
[  715.272975] usb 1-1: USB disconnect, device number 2
[  724.928574] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
[  734.967644] usb 2-1: USB disconnect, device number 2
[  735.487717] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
[  735.522827] rndis_host 2-1:1.0 eth2: register 'rndis_host' at usb-11200000.usb-1, ZTE RNDIS device, 10:3c:59:de:f1:f1
[  738.998773] rndis_host 2-1:1.0 eth2: entered allmulticast mode
[  738.999112] br-lan: left allmulticast mode
[  739.077438] br-lan: entered allmulticast mode
[  795.512783] usb 2-1: USB disconnect, device number 3
[  795.512954] rndis_host 2-1:1.0 eth2: unregister 'rndis_host' usb-11200000.usb-1, ZTE RNDIS device
[  795.733488] rndis_host 2-1:1.0 eth2 (unregistering): left allmulticast mode
[  798.512990] usb 2-1: new SuperSpeed USB device number 4 using xhci-mtk
[  798.549383] rndis_host 2-1:1.0 eth2: register 'rndis_host' at usb-11200000.usb-1, ZTE RNDIS device, 10:3c:59:de:f1:f1
[  799.342480] usb 2-1: USB disconnect, device number 4
[  799.342632] xhci-mtk 11200000.usb: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  799.342658] rndis_host 2-1:1.0 eth2: unregister 'rndis_host' usb-11200000.usb-1, ZTE RNDIS device
[  799.342868] xhci-mtk 11200000.usb: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[  801.602332] xhci-mtk 11200000.usb: ERROR: unexpected setup address command completion code 0x11.
[  801.822307] xhci-mtk 11200000.usb: ERROR: unexpected setup address command completion code 0x11.
[  802.042259] usb 2-1: device not accepting address 5, error -22
[  802.282274] xhci-mtk 11200000.usb: ERROR: unexpected setup address command completion code 0x11.
[  802.502248] xhci-mtk 11200000.usb: ERROR: unexpected setup address command completion code 0x11.
[  802.722219] usb 2-1: device not accepting address 6, error -22
[  802.722357] usb usb2-port1: attempt power cycle
[  811.751755] usb usb2-port1: unable to enumerate USB device
[  812.731444] usb 2-1: new SuperSpeed USB device number 9 using xhci-mtk
[  817.971282] rndis_host 2-1:1.0: RNDIS init failed, -19
[  818.381019] usb 2-1: USB disconnect, device number 9
[... 244 lines cut ...]
### ImmortalWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1 (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Steps to reproduce

接入F50启动完成 添加USB WAN接口 指定设备为eth2 接口获取到ip 接口断开 eth2设备不存在 F50并未自动重启 过程重复：接口获取到ip 接口断开 eth2设备不存在 接口获取到ip 接口断开 eth2设备不存在。。。。

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

padavanonly, 2025-03-19:
没有f50,有这个的来搞吧

libriunk, 2025-03-19:
这玩意看批次25年之前的需要改供电。25年批次更新了供电，但仍需关闭f50的wifi功能使用

lylus, 2025-03-20:
前几天某东新买的，也是刷成24.10固件，就一直出现和上面说的一模一样的问题，单单供电模块和JUSB线，就买了过百元。4-5条usb3.0各种原装线和双路供电等，usb 5v3a模块、5v5A模块，依然没有解决，就是出现上述问题，表现为F50没有重启，但是24.10接口-设备里面eth2成功显示1-2秒就消失，再次显示，再次消失。最后选择退货。用回CMCC RAX3000 23.05的固件很稳。RAX3000如果刷24.10也有问题的，第一次上电会检测不到F50，要等RAX3000启动完成后再重启一下F50才正常，而23.05则没有此问题。

libriunk, 2025-03-20:
> 前几天某东新买的，也是刷成24.10固件，就一直出现和上面说的一模一样的问题，单单供电模块和JUSB线，就买了过百元。4-5条usb3.0各种原装线和双路供电等，usb 5v3a模块、5v5A模块，依然没有解决，就是出现上述问题，表现为F50没有重启，但是24.10接口-设备里面eth2成功显示1-2秒就消失，再次显示，再次消失。最后选择退货。用回CMCC RAX3000 23.05的固件很稳。RAX3000如果刷24.10也有问题的，第一次上电会检测不到F50，要等RAX3000启动完成后再重启一下F50才正常，而23.05则没有此问题。

所以您f50的无线关了吗？ 这玩意供电有问题 换线是没用的

lylus, 2025-03-20:
> 所以您f50的无线关了吗？ 这玩意供电有问题 换线是没用的

关了的，也锁了频，我个人感觉问题大概率是在TR3000的硬件上面，它的usb也没有做屏蔽，我用在rax3000上，就可以正常使用。

h2ch, 2025-03-25:
> 供电不够导致的。

从C口电源飞线过去后供电够，但是在工作20分钟左右就会出现这几条日志，然后断网

Tue Mar 25 06:56:20 2025 daemon.info procd: Instance odhcpd::instance1 pid 2421 not stopped on SIGTERM, sending SIGKILL instead
Tue Mar 25 06:56:20 2025 kern.info kernel: [  822.604631] br-lan: left allmulticast mode
Tue Mar 25 06:56:20 2025 kern.info kernel: [  822.731677] cdc_ether 2-1:1.0 eth2: entered allmulticast mode
Tue Mar 25 06:56:21 2025 kern.info kernel: [  822.811517] br-lan: entered allmulticast mode
Tue Mar 25 06:56:33 2025 authpriv.info dropbear[8635]: Exit (root) from <172.16.1.115:61545>: Disconnect received
Tue Mar 25 07:03:49 2025 daemon.notice netifd: USBWAN (14957): udhcpc: lease of 172.16.0.101 obtained from 172.16.0.1, lease time 43199
Tue Mar 25 07:03:49 2025 daemon.notice netifd: Interface 'USBWAN' is now up
Tue Mar 25 07:03:49 2025 user.notice firewall: Reloading firewall due to ifup of USBWAN (eth2)
Tue Mar 25 07:03:50 2025 daemon.err odhcp6c[14965]: Failed to send RS (Address not available)
Tue Mar 25 07:03:50 2025 daemon.err odhcp6c[14965]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Mar 25 07:03:54 2025 daemon.err odhcp6c[14965]: Failed to send RS (Address not available)
Tue Mar 25 07:03:54 2025 daemon.err odhcp6c[14965]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Mar 25 07:03:58 2025 daemon.err odhcp6c[14965]: Failed to send RS (Address not available)
Tue Mar 25 07:04:03 2025 daemon.err odhcp6c[14965]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Mar 25 07:04:22 2025 daemon.err odhcp6c[14965]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Mar 25 07:05:02 2025 daemon.err odhcp6c[14965]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Mar 25 07:06:29 2025 daemon.err odhcp6c[14965]: Failed to send SOLICIT message to ff02::1:2 (Address not available)

内核日志：

[  380.606383] cdc_ether 2-1:1.0 eth2: register 'cdc_ether' at usb-11200000.usb-1, ZTE CDC Ethernet Device, 10:3c:59:de:f1:f1
[  822.604631] br-lan: left allmulticast mode
[  822.731677] cdc_ether 2-1:1.0 eth2: entered allmulticast mode
[  822.811517] br-lan: entered allmulticast mode
[ 1202.947327] br-lan: left allmulticast mode
[ 1203.046901] br-lan: entered allmulticast mode
[ 1203.217059] br-lan: left allmulticast mode
[ 1203.306848] br-lan: entered allmulticast mode
[ 1236.720549] usb 2-1: USB disconnect, device number 5
[ 1236.720719] cdc_ether 2-1:1.0 eth2: unregister 'cdc_ether' usb-11200000.usb-1, ZTE CDC Ethernet Device
[ 1236.731412] cdc_ether 2-1:1.0 eth2 (unregistering): left allmulticast mode
[ 1243.199366] usb 1-1: new full-speed USB device number 3 using xhci-mtk
[ 1243.389523] usb 1-1: not running at top speed; connect to a high speed hub
[ 1247.635731] usb 1-1: USB disconnect, device number 3
[ 1257.287048] usb 2-1: new SuperSpeed USB device number 6 using xhci-mtk
[ 1267.215322] usb 2-1: USB disconnect, device number 6
[ 1267.735337] usb 2-1: new SuperSpeed USB device number 7 using xhci-mtk
[ 1267.769809] cdc_ether 2-1:1.0 eth2: register 'cdc_ether' at usb-11200000.usb-1, ZTE CDC Ethernet Device, 10:3c:59:de:f1:f1
[ 1269.210603] IPv6: eth2: IPv6 duplicate address fe80::123c:59ff:fede:f1f1 used by 10:3c:59:de:f1:f1 detected!

h2ch, 2025-03-25:
拔掉F50电源硬重启后，odhcpd服务会占用50%-60% CPU，手动重启才正常。

[image]

root@ImmortalWrt:~# logread -e odhcpd
Tue Mar 25 06:42:28 2025 user.notice ucitrack: Setting up /etc/config/odhcpd reload dependency on /etc/config/dhcp
Tue Mar 25 06:42:29 2025 daemon.err odhcpd[2421]: Failed to send to ff02::2%USBWAN@eth2 (Address not available)
Tue Mar 25 06:42:31 2025 daemon.err odhcpd[2421]: Failed to send to ff02::1%lan@br-lan (Address not available)
Tue Mar 25 06:42:31 2025 daemon.err odhcpd[2421]: Failed to send to ff02::1%lan@br-lan (Address not available)
Tue Mar 25 06:48:18 2025 daemon.warn odhcpd[2421]: No default route present, setting ra_lifetime to 0!
Tue Mar 25 06:48:18 2025 daemon.err odhcpd[2421]: Failed to send to ff02::1%USBWAN6@eth2 (No such device)
Tue Mar 25 06:56:20 2025 daemon.info procd: Instance odhcpd::instance1 pid 2421 not stopped on SIGTERM, sending SIGKILL instead
Tue Mar 25 07:03:15 2025 daemon.warn odhcpd[8981]: No default route present, setting ra_lifetime to 0!
Tue Mar 25 07:03:15 2025 daemon.err odhcpd[8981]: Failed to send to ff02::1%USBWAN6@eth2 (No such device)
Tue Mar 25 07:13:34 2025 daemon.info procd: Instance odhcpd::instance1 pid 8981 not stopped on SIGTERM, sending SIGKILL instead
Tue Mar 25 07:13:35 2025 daemon.err odhcpd[16273]: Failed to send to ff02::2%USBWAN@eth2 (Address not available)

mia0x75, 2025-03-26:
@h2ch 请问，这个路由怎么刷入immortalwrt呢？我使用了24.10.0这个标签的源码，自己挑了一些app和utility，折腾好多天，可以编出来一个70兆左右的包，但刷入后无法启动。已经刷好了恩山的三分区uboot，把编译好的固件在uboot刷入112/114，或者在别人的openwrt上升级自己编译的固件都是无法启动，现象是重启后白灯亮起稍后闪烁在关闭，然后再亮起闪烁关闭。。。。

h2ch, 2025-03-26:
> [@h2ch](https://github.com/h2ch) 请问，这个路由怎么刷入immortalwrt呢？我使用了24.10.0这个标签的源码，自己挑了一些app和utility，折腾好多天，可以编出来一个70兆左右的包，但刷入后无法启动。已经刷好了恩山的三分区uboot，把编译好的固件在uboot刷入112/114，或者在别人的openwrt上升级自己编译的固件都是无法启动，现象是重启后白灯亮起稍后闪烁在关闭，然后再亮起闪烁关闭。。。。

用immortalwrt预编译镜像，用imm官方uboot和bl2或者某山捅不死uboot配合imm bl2刷入default分区，建议用捅不死，imm uboot需要镜像名符合规则，也不带web

moqingzhao, 2025-05-12:
大佬们，我编译固件为什么无法获取ip啊，默认的配置

[image]
是设置好固件型号后，在menuconfig是 还要勾选哪个插件吗？困扰我好几天了，求大佬指教

moqingzhao, 2025-05-12:
> [@h2ch](https://github.com/h2ch) 请问，这个路由怎么刷入immortalwrt呢？我使用了24.10.0这个标签的源码，自己挑了一些app和utility，折腾好多天，可以编出来一个70兆左右的包，但刷入后无法启动。已经刷好了恩山的三分区uboot，把编译好的固件在uboot刷入112/114，或者在别人的openwrt上升级自己编译的固件都是无法启动，现象是重启后白灯亮起稍后闪烁在关闭，然后再亮起闪烁关闭。。。。

大佬解决了吗

1715173329, 2025-05-12:
请不要在本 issue 下发布无关的回复。

haoyouab, 2026-02-21:
> 666666内核问题都来了 在f50(已root、已装载128g tf、开启wifi未连接设备、性能模式、中国广电n78、qci9)通过USB3.0 A2C数据线连接至5v3a适配器供电的tr3000(ubootmod)使用今年1月份的snapshot，使用手机链接tr3000无线(握手2401)、使用全球网测app测速，结果是300/50，下载6g的大文件时未发现出现断流/重启的情况。

immortal wrt的snapshot吗？固件下载页面现在是不是挂掉了访问不了？

haoyouab, 2026-02-22:
> 这是内核问题，无解，目前在用的某op固件6.6.63版本内核完全正常，高于低于这版本的都不行，要么试试自行编译这个版本的内核，要么乖乖转网口
> 
> Sent from [Proton Mail](https://proton.me/mail/home) for iOS.
> 
> -------- Original Message --------
> 
> > haoyouab left a comment [(immortalwrt/immortalwrt#1764)](https://github.com/immortalwrt/immortalwrt/issues/1764#issuecomment-3939000758)
> >
> > 我也遇到类似的情况，没找到解决办法

可以分享一下固件吗，感谢！我邮箱haoyouab@gmail.com