immortalwrt/immortalwrt #1782: pptpd 服务器使用 外面客户端连接提示  LCP terminated by peer (MPPE required but not available)

tian4546, 2025-03-29:
### Describe the bug

Sat Mar 29 13:08:32 2025 daemon.info pptpd[18434]: CTRL: Client 116.162.227.30 control connection started
Sat Mar 29 13:08:33 2025 daemon.info pptpd[18434]: CTRL: Starting call (launching pppd, opening GRE)
Sat Mar 29 13:08:33 2025 daemon.notice pppd[18448]: pppd 2.5.1 started by root, uid 0
Sat Mar 29 13:08:33 2025 daemon.info pppd[18448]: Using interface ppp1
Sat Mar 29 13:08:33 2025 daemon.notice pppd[18448]: Connect: ppp1 <--> /dev/pts/2
Sat Mar 29 13:08:33 2025 daemon.notice pppd[18448]: peer from calling number 116.162.227.30 authorized
Sat Mar 29 13:08:33 2025 daemon.info pppd[18448]: LCP terminated by peer (MPPE required but not available)
Sat Mar 29 13:08:33 2025 daemon.debug pptpd[18434]: CTRL: Reaping child PPP[18448]
Sat Mar 29 13:08:33 2025 daemon.notice pppd[18448]: Modem hangup
Sat Mar 29 13:08:33 2025 daemon.notice pppd[18448]: Connection terminated.
Sat Mar 29 13:08:33 2025 daemon.info pppd[18448]: Exit.
Sat Mar 29 13:08:33 2025 daemon.info pptpd[18434]: CTRL: Client 116.162.227.30 control connection finished
Sat Mar 29 13:08:59 2025 authpriv.info dropbear[12740]: Exit (root) from <192.168.2.1:41378>: Exited normally

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

安装pptpd 配置外面连接就这个错我

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

tian4546, 2025-03-29:
ai 回复编译 ppp 这个插件的时候应该 ./configure --enable-mppe 启用mppe  

[image]   

pppd --help | grep mppe 这个命令可以查看是否支持 mppe加密

syhzjnbxs, 2025-05-16:
问题有解决了么？24.10以后的openwrt，pptpd服务器不能用了，也是相同的错误

tian4546, 2025-05-16:
一直没有，需要作者调试

closed 2025-05-19 by 1715173329