immortalwrt/immortalwrt #1796: passwall 不能正常安装使用

404area, 2025-04-05:
### Describe the bug

缺少依赖

### ImmortalWrt version

23.0.x

### ImmortalWrt release

r28061-399f9a1db3

### ImmortalWrt target/subtarget

armsr/armv8

### Device

passwall

### Image kind

Official downloaded image

### Steps to reproduce

安装passwall 总是提示依赖的软包 shadowsocks-rust-sslocal在所在的仓库都未提供，所有的23.x版本都是，包括docker镜像和虚拟磁盘镜像，只要是新安装就提示没有这个依赖

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

libiunc, 2025-04-05:
强制安装 passwall 然后插件选singbox 基本上就够了。您的设备是什么？

404area, 2025-04-05:
> 强制安装 passwall 然后插件选singbox 基本上就够了。您的设备是什么？

只要安装就会提示没有这个依赖，就会报错，无法强制安装，之前没遇到过，这次重新安装就遇到了，在虚拟机安装的，直接用的磁盘映像