immortalwrt/immortalwrt #1833: 编译时 提示补丁错误

x1906875173, 2025-04-23:
### Describe the bug

patching file net/ipv4/tcp_offload.c
Hunk #1 FAILED at 367.
1 out of 1 hunk FAILED -- saving rejects to file net/ipv4/tcp_offload.c.rej
Patch failed!  Please fix immortalwrt/target/linux/generic/pending-6.6/690-net-add-missing-check-for-TCP-fraglist-GRO.patch!

### ImmortalWrt version

最新

### ImmortalWrt release

快照

### ImmortalWrt target/subtarget

x86/64

### Device

N5000

### Image kind

Official downloaded image

### Steps to reproduce

编译时 提示补丁错误 

### Actual behaviour

编译时 提示补丁错误 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

x1906875173, 2025-04-23:
Applying /home/ues-t/test/immortalwrt/target/linux/generic/pending-6.6/690-net-add-missing-check-for-TCP-fraglist-GRO.patch using plaintext: 
patching file net/ipv4/tcp_offload.c
Hunk #1 FAILED at 367.
1 out of 1 hunk FAILED -- saving rejects to file net/ipv4/tcp_offload.c.rej
Patch failed!  Please fix /home/ues-t/test/immortalwrt/target/linux/generic/pending-6.6/690-net-add-missing-check-for-TCP-fraglist-GRO.patch!
make[4]: *** [Makefile:35: /home/ues-t/test/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.87/.prepared_d91803146fb4df2a2262f36c03b5c50d] Error 1
make[4]: Leaving directory '/home/ues-t/test/immortalwrt/target/linux/x86'
make[3]: *** [Makefile:12: compile] Error 2
make[3]: Leaving directory '/home/ues-t/test/immortalwrt/target/linux'
time: target/linux/compile#6.53#2.72#2.71
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/ues-t/test/immortalwrt'
make[1]: *** [target/Makefile:25: /home/ues-t/test/immortalwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ues-t/test/immortalwrt'
make: *** [/home/ues-t/test/immortalwrt/include/toplevel.mk:233：world] 错误 2

initzhu, 2025-04-23:
等合并上游源码，op已经修了