immortalwrt/immortalwrt #1837: 24.10.1刷完仍然不能进后台

johnsmith8736, 2025-04-25:
### Describe the bug

24.10.1刷完仍然不能进后台

### ImmortalWrt version

24.10.1

### ImmortalWrt release

24.10.1

### ImmortalWrt target/subtarget

VMware 

### Device

x86/64

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

MarksonHon, 2025-04-25:
@johnsmith8736 VMware 虚拟网卡配置的截图呢？

johnsmith8736, 2025-04-25:
> [@johnsmith8736](https://github.com/johnsmith8736) VMware 虚拟网卡配置的截图呢？

[image]

miyatsu, 2025-04-25:
I've download the latest release `immortalwrt-24.10.1-x86-64-generic-squashfs-combined-efi.vhdx`, booting in Hyper-V enviourment (`Windows Server 2025` as host) without any issue. Both `web (luci)` and `ssh` work as expected.

[image]

johnsmith8736, 2025-04-26:
> I've download the latest release `immortalwrt-24.10.1-x86-64-generic-squashfs-combined-efi.vhdx`, booting in Hyper-V enviourment (`Windows Server 2025` as host) without any issue. Both `web (luci)` and `ssh` work as expected.
> 
> [image]

我用的是这个COMBINED-EFI (SQUASHFS-COMBINED-EFI.IMG.GZ)用StarWind V2V Converter / P2V Migrator 转成vmdk文件的。
I used this COMBINED-EFI (SQUASHFS-COMBINED-EFI.IMG.GZ) with StarWind V2V Converter / P2V Migrator to convert to vmdk file.

johnsmith8736, 2025-04-26:
直接用vmdk 文件虽然当时行，下次启动时就报错了，不能启动。
Although the vmdk file works at the time, the next time you start it, you get an error and it won't start.

warmJN, 2025-04-29:
你有几个网口呢，如果是四个，我看设备列表里只有三个，会不会刚好网线插在了不在列表的网口

johnsmith8736, 2025-04-29:
> 你有几个网口呢，如果是四个，我看设备列表里只有三个，会不会刚好网线插在了不在列表的网口

4个网口，eth0,eth1,eth2是lan口，eth3是wan口

miyatsu, 2025-04-29:
> 你有几个网口呢，如果是四个，我看设备列表里只有三个，会不会刚好网线插在了不在列表的网口

Sounds reasonable. I've encountered with such issue before.

I have a VM with multiple nic, the interface name assigned to each nic was not sequential aligned with real hardware (in vm manager panal). It will works like this:

```
nic1, nic2, nic3

1st boot:
nic1->eth0 (This should be the default lan interface)
nic2->eth1 (This should be the default wan interface)
nic3->eth2
...

2nd/3rd/4th boot:
nic2->eth0 (Lan now bridge to nic2)
nic3->eth1
nic1->eth2 (Original lan assigne to eth2, which not present in /etc/config/network by default)
```

It seems that the Linux kernel assign ethX name to nic with the order of discovering them. The first detected nic will be assigned with name `eth0`, and so on. I've tested this multiple times, it may result in `nic2` assigned with name `eth0` at 1st boot.

I've write an `rc` script put to `/etc/rc.d/` directory to rename nic name to `ethXX` where `XX` is the last byte of nic's `mac address` to fix this issue.

```
#!/bin/sh

# Get all network device
NET_DEVS="$(ls /sys/class/net/*/device/uevent | awk -F '/' '{print $5}')"

# Loop all network device
for NET_DEV0 in ${NET_DEVS}
do
	# Rename network device name to mac address like this: eth01 eth11 etc.
	NEW_NAME="eth$(cat /sys/class/net/"${NET_DEV0}"/address | awk -F 'FILL with MAC prefix' '{print $1}')"

	ip link set "${NET_DEV0}" down
	ip link set "${NET_DEV0}" name "${NEW_NAME}"
	ip link set "${NEW_NAME}" up
done

# After change eth0 to eth11, set ports to eth11 in order make lan network works
# uci set network.@device[0].ports='eth11'
# uci commit

# Reload
/etc/init.d/network reload
```

miyatsu, 2025-04-29:
> 直接用vmdk 文件虽然当时行，下次启动时就报错了，不能启动。 Although the vmdk file works at the time, the next time you start it, you get an error and it won't start.

Try to assign only one nic1 to your vm, only eth0 (nic1, lan) present.

johnsmith8736, 2025-07-24:
可能不是immortalwrt固件的问题，我昨天换成了PVE,成功启动了[24.10.2],进了后台！