immortalwrt/immortalwrt #1839: 24.10.1无法正常启动

lhq46688, 2025-04-25:
### Describe the bug

在PE环境下使用写盘工具写入镜像，启动后一直卡在gnu grub界面，需要按enter后可以启动，并且后续启动都要按一下，尝试修改/boot/grub/grub.cfg，set timeout为0后可以正常启动，其他未作修改
bios为默认uefi，磁盘分区表为gpt使用镜像为immortalwrt-24.10.1-x86-64-generic-squashfs-combined.img

### ImmortalWrt version

24.10.1

### ImmortalWrt release

immortalwrt-24.10.1-x86-64-generic-squashfs-combined.img

### ImmortalWrt target/subtarget

x86/64

### Device

cpu型号n150的x86-64小主机

### Image kind

Official downloaded image

### Additional info

root@ImmortalWrt:~# cat /boot/grub/grub.cfg
serial --unit=0 --speed=115200 --word=8 --parity=no --stop=1 --rtscts=off
terminal_input console serial; terminal_output console serial

set default="0"
set timeout="0"
search -l kernel -s root

menuentry "ImmortalWrt" {
        linux /boot/vmlinuz root=PARTUUID=ab4df169-d914-20f2-9f59-24a03fbdf802 r                                                                                                                                                                                                                                             ootwait   console=tty1 console=ttyS0,115200n8 noinitrd
}
menuentry "ImmortalWrt (failsafe)" {
        linux /boot/vmlinuz failsafe=true root=PARTUUID=ab4df169-d914-20f2-9f59-                                                                                                                                                                                                                                             24a03fbdf802 rootwait   console=tty1 console=ttyS0,115200n8 noinitrd
}
root@ImmortalWrt:~#

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-04-25:
https://github.com/immortalwrt/immortalwrt/discussions/1366#discussioncomment-12344060

lhq46688, 2025-04-25:
谢谢你的回复，如果是需要通过编辑grub.cfg，删除terminal_input console serial的话，和我现在更改set timeout为0没有什么区别，我希望找到方法可以解决刷写全新系统时不会遇到同样的问题，谢谢

closed 2025-04-25 by lhq46688