immortalwrt/immortalwrt #1840: cpu占用持续保持在27左右无法下降

shjy10000, 2025-04-25:
### Describe the bug

cpu占用持续保持在27左右无法下降，经过排查应该是显示错误而已

### ImmortalWrt version

r33758-b3a95ad3c7

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Radxa E25 Carrier Board

### Image kind

Official downloaded image

### Steps to reproduce

cpu占用持续保持在27左右无法下降

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

xlighting2017, 2025-04-26:
截图没有， 日志没有， 然后就 「经过排查」

那只能说： 我这里是好的， 你的显示器坏了

另外， https://downloads.immortalwrt.org/snapshots/targets/rockchip/armv8/  这里只有 33717,

你「Official downloaded image」 哪来的 33758 ？

shjy10000, 2025-04-26:
> 截图没有， 日志没有， 然后就 「经过排查」
> 
> 那只能说： 我这里是好的， 你的显示器坏了
> 
> 另外， https://downloads.immortalwrt.org/snapshots/targets/rockchip/armv8/ 这里只有 33717，
> 
> 你「官方下载的图片」 哪来的 33758 ？

上面所有的版本信息来自命令，经过排查是因为，这个持续的占用不影响日常用网，温度和频率都没有变化，我从从新贴上日志和截图吧，

shjy10000, 2025-04-26:
> > 截图没有， 日志没有， 然后就 「经过排查」
> > 那只能说： 我这里是好的， 你的显示器坏了
> > 另外， https://downloads.immortalwrt.org/snapshots/targets/rockchip/armv8/ 这里只有 33717，
> > 你「官方下载的图片」 哪来的 33758 ？
> 
> 上面所有的版本信息来自命令，经过排查是因为，这个持续的占用不影响日常用网，温度和频率都没有变化，我从从新贴上日志和截图吧，

[image]

shjy10000, 2025-04-26:
> > 截图没有， 日志没有， 然后就 「经过排查」
> > 那只能说： 我这里是好的， 你的显示器坏了
> > 另外， https://downloads.immortalwrt.org/snapshots/targets/rockchip/armv8/ 这里只有 33717，
> > 你「官方下载的图片」 哪来的 33758 ？
> 
> 上面所有的版本信息来自命令，经过排查是因为，这个持续的占用不影响日常用网，温度和频率都没有变化，我从从新贴上日志和截图吧，

Sat Apr 26 05:21:06 2025 user.notice nft-qos-dynamic: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01
Sat Apr 26 05:21:06 2025 user.notice nft-qos-dynamic: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01
Sat Apr 26 05:21:11 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: started, version 2.91 cachesize 8000
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth DNSSEC no-ID loop-detect inotify dumpfile
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 3d
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using nameserver 211.140.13.188#53
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using nameserver 211.140.188.188#53
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using nameserver 2409:8028:2000::1111#53
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using nameserver 2409:8028:2000::2222#53
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Apr 26 05:21:12 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Apr 26 05:21:12 2025 user.notice nft-qos-monitor: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01
Sat Apr 26 05:21:12 2025 user.notice nft-qos-monitor: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01
Sat Apr 26 05:21:12 2025 user.notice nft-qos-dynamic: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01
Sat Apr 26 05:21:12 2025 user.notice nft-qos-dynamic: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01
Sat Apr 26 05:22:05 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: started, version 2.91 cachesize 8000
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth DNSSEC no-ID loop-detect inotify dumpfile
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 3d
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using nameserver 211.140.13.188#53
Sat Apr 26 05:22:06 2025 daemon.info dnsmasq[1]: using nameserver 211.140.188.188#53
[... cut ...]
Sat Apr 26 07:22:42 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat Apr 26 07:22:42 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Apr 26 07:22:42 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Sat Apr 26 07:22:42 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Apr 26 07:22:42 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Apr 26 07:22:43 2025 user.notice nft-qos-monitor: ACTION=update, MACADDR=50:4f:3b:4c:fc:a4, IPADDR=192.168.10.246, HOSTNAME=MiWiFi-RC01

shjy10000, 2025-04-26:
你说的版本不对我不知道什么原因，以上所有的版本信息，都是使用的反馈里面的命令获取的，使用的都是官方源码编译的，问题就是图片里面的那个cpu占用会议一直保持在27作用，如果测速，会到44到50 作用，我觉得是显示错误的原因是因为虽然占用一直很高，但是网络不卡，温度也没有任何变高，频率也没有任何的变化

shjy10000, 2025-04-26:
> > 截图没有， 日志没有， 然后就 「经过排查」
> > 那只能说： 我这里是好的， 你的显示器坏了
> > 另外， https://downloads.immortalwrt.org/snapshots/targets/rockchip/armv8/ 这里只有 33717，
> > 你「官方下载的图片」 哪来的 33758 ？
> 
> 上面所有的版本信息来自命令，经过排查是因为，这个持续的占用不影响日常用网，温度和频率都没有变化，我从从新贴上日志和截图吧，

你说的版本不对我不知道什么原因，以上所有的版本信息，都是使用的反馈里面的命令获取的，使用的都是官方源码编译的，问题就是图片里面的那个cpu占用会议一直保持在27作用，如果测速，会到44到50 作用，我觉得是显示错误的原因是因为虽然占用一直很高，但是网络不卡，温度也没有任何变高，频率也没有任何的变化，使用24.10分支编译也是会有这个问题，而且在之前的版本就一直存在这个问题，无论是用哪个分支源码编译

shjy10000, 2025-04-26:
> > 截图没有， 日志没有， 然后就 「经过排查」
> > 那只能说： 我这里是好的， 你的显示器坏了
> > 另外， https://downloads.immortalwrt.org/snapshots/targets/rockchip/armv8/ 这里只有 33717，
> > 你「官方下载的图片」 哪来的 33758 ？
> 
> 上面所有的版本信息来自命令，经过排查是因为，这个持续的占用不影响日常用网，温度和频率都没有变化，我从从新贴上日志和截图吧，

这是内核日志
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.87 (sh@sh-virtual-machine) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r33758-b3a95ad3c7) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Fri Apr 18 14:08:05 2025
[    0.000000] KASLR enabled
[    0.000000] Machine model: Radxa E25 Carrier Board
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x000000007ba00000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 19 pages/cpu s38696 r8192 d30936 u77824
[    0.000000] pcpu-alloc: s38696 r8192 d30936 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyS2,115200 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 515592
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007d600000-0x000000007d800000] (2MB)
[    0.000000] Memory: 2017712K/2095104K available (9664K kernel code, 2454K rwdata, 2936K rodata, 2240K init, 734K bss, 61008K reserved, 16384K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[    0.000000] GIC: enabling workaround for ITS: Rockchip erratum RK3568001
[    0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @450000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @460000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000000470000
[    0.000000] GIC: using cache flushing for LPI property table
[... 447 lines cut ...]
[  269.357260] r8125 0000:01:00.0 eth0: left promiscuous mode
[  269.363636] br-lan: port 1(eth0) entered disabled state
[  269.526254] eth0: 0xffff8000806e0000, 22:41:7e:32:1e:6a, IRQ 63
[  269.545360] br-lan: port 1(eth0) entered blocking state
[  269.551276] br-lan: port 1(eth0) entered disabled state
[  269.557183] r8125 0000:01:00.0 eth0: entered allmulticast mode
[  269.563912] r8125 0000:01:00.0 eth0: entered promiscuous mode
[  271.851483] r8125: eth0: Link is Up - 2.5Gbps/Full
[  271.856921] br-lan: port 1(eth0) entered blocking state
[  271.862773] br-lan: port 1(eth0) entered forwarding state
[  350.931999] br-lan: port 1(eth0) entered disabled state
[  350.939186] r8125 0000:01:00.0 eth0: left allmulticast mode
[  350.945535] r8125 0000:01:00.0 eth0: left promiscuous mode
[  350.951837] br-lan: port 1(eth0) entered disabled state
[  351.081514] eth0: 0xffff8000806e0000, 22:41:7e:32:1e:6a, IRQ 63
[... cut ...]
[  351.107253] br-lan: port 1(eth0) entered disabled state
[  351.113145] r8125 0000:01:00.0 eth0: entered allmulticast mode
[  351.119872] r8125 0000:01:00.0 eth0: entered promiscuous mode
[  351.127565] br-lan: port 1(eth0) entered blocking state
[  351.133428] br-lan: port 1(eth0) entered forwarding state
[  351.949908] br-lan: port 1(eth0) entered disabled state
[  353.407478] r8125: eth0: Link is Up - 2.5Gbps/Full
[  353.412903] br-lan: port 1(eth0) entered blocking state
[  353.418753] br-lan: port 1(eth0) entered forwarding state