immortalwrt/immortalwrt #1843: 24.10.1版本在ZTE E8820s上无法启动

jerry-yuan, 2025-04-26:
### Describe the bug

按24.10.1的默认配置自己编译了一个版本,刷上去以后无法启动.

看TTL的错误提示,应该是正确找到了内核但是找不到rootfs,导致Kernel Panic挂掉了:
```
[    4.622870] clk: Disabling unused clocks
[    4.627300] MTD: Couldn't look up '': -22
[    4.631416] /dev/root: Can't open blockdev
[    4.635550] VFS: Cannot open root device "" or unknown-block(0,0): error -6
[    4.642482] Please append a correct "root=" boot option; here are the available partitions:
[    4.650830] 1f00            2176 mtdblock0 
[    4.650844]  (driver?)
[    4.657364] 1f01            1280 mtdblock1 
[    4.657375]  (driver?)
[    4.663916] 1f02            1280 mtdblock2 
[    4.663927]  (driver?)
[    4.670427] 1f03            4096 mtdblock3 
[    4.670435]  (driver?)
[    4.676956] 1f04          121728 mtdblock4 
[    4.676967]  (driver?)
[    4.683491] List of all bdev filesystems:
[    4.687480]  squashfs
[    4.687486] 
[    4.691218] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    4.699460] Rebooting in 1 seconds..
```
强制通过bootargs指定mtdblock4,又发现它在尝试使用jffs的函数去读取squashfs的样子,不停的报错Magic Number不匹配.

暂未尝试master最新版本代码是否有相同问题.

### ImmortalWrt version

v24.10.1

### ImmortalWrt release

v24.10.1

### ImmortalWrt target/subtarget

ramips/zte_e8820s

### Device

ZTE E8820s

### Image kind

Self-built image

### Steps to reproduce

检出v24.10.1 Tag,选择ramips/zte_e8820s目标，按默认配置直接编译下载

### Actual behaviour

TTL报错如下,因找不到rootfs导致Kernel Panic
```
[    4.622870] clk: Disabling unused clocks
[    4.627300] MTD: Couldn't look up '': -22
[    4.631416] /dev/root: Can't open blockdev
[    4.635550] VFS: Cannot open root device "" or unknown-block(0,0): error -6
[    4.642482] Please append a correct "root=" boot option; here are the available partitions:
[    4.650830] 1f00            2176 mtdblock0 
[    4.650844]  (driver?)
[    4.657364] 1f01            1280 mtdblock1 
[    4.657375]  (driver?)
[    4.663916] 1f02            1280 mtdblock2 
[    4.663927]  (driver?)
[    4.670427] 1f03            4096 mtdblock3 
[    4.670435]  (driver?)
[    4.676956] 1f04          121728 mtdblock4 
[    4.676967]  (driver?)
[    4.683491] List of all bdev filesystems:
[    4.687480]  squashfs
[    4.687486] 
[    4.691218] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    4.699460] Rebooting in 1 seconds..
```

### Expected behaviour

正常启动

### Additional info

在dts中,bootargs指定root=/dev/mtdblock4 rootfstype=squashfs,发现启动后似乎在尝试用jffs解析squashfs分区,报错MagicNumber不匹配.引入kmod-fs-squashfs仍然无解.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-04-26:
需要刷本人的 uboot，具体看历史提交记录。

jerry-yuan, 2025-04-26:
> 需要刷本人的 uboot，具体看历史提交记录。

好的,感谢.想了解一下这里边的原理,请问有什么推荐的文档可以参考吗?

1715173329, 2025-04-26:
分区表不同而已

closed 2025-04-26 by 1715173329

jerry-yuan, 2025-04-26:
> 分区表不同而已

感谢大佬回复,从这个点出发,看了下我用的项目里边的dts和immortalwrt的dts中定义的分区确实不太一样.剩下的我自己再研究学习一下吧.