immortalwrt/immortalwrt #1845: redmi ax6s 用一段时间之后，通过wifi来复制分享nas（nas接的有线口）会网络错误，无法进行复制、读取、分享，内网异常。但是设备上外网是正常的（附日志），初步怀疑是运营商isp强制下线重播导致的？

sss132919, 2025-04-28:
### Describe the bug

[ax6s-immoralwrt日志.txt](https://github.com/user-attachments/files/19934219/ax6s-immoralwrt.txt)

### ImmortalWrt version

ImmortalWrt 24.10.0 r32824-6a73dae98c9c / LuCI openwrt-24.10 branch 25.017.24510~d42ec55

### ImmortalWrt release

ImmortalWrt 24.10.0 r32824-6a73dae98c9c / LuCI openwrt-24.10 branch 25.017.24510~d42ec55

### ImmortalWrt target/subtarget

Xiaomi Redmi Router AX6S

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Actual behaviour

redmi ax6s 用一段时间之后，通过wifi来复制分享nas（nas接的有线口）会网络错误，无法进行复制、读取、分享，内网异常。但是设备上外网是正常的（附日志），初步怀疑是运营商isp强制下线重播导致的？

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

kmahyyg, 2025-05-09:
不是，这个问题在 23.05.4 是没有的，24.10 开始才有，24.10.1 我不知道有没有。问题似乎是在 MTK 的 WED 驱动 的 connection tracking 上，仅在 无线->有线 这种情况才有。稳定出问题的表现就是 内网无线到内网有线的 上传速度永远不足  10mbps。
我之前在用其他人的 23.x/22.x 也是有这个问题，只有 immortalwrt 的 23.05 是相对正常的。
如果24.10.1 没有解决，建议临时回退 23.05.4 可破，回退前做好备份，由于固件布局变更，请以救砖流程回退官方固件后刷机。

kmahyyg, 2025-05-09:
To Maintainer: 我不是专业开发者，从用户角度看，最明显的问题在不正常的版本看上去就是 /sys/kernel/debug/ppe0/entries 下面有非常多（好几屏，此时网络几乎静默，约等于没有流量）UNB ，而且不会自动清除。重启后可短暂恢复使用。
在正常的版本下 entries 下的内容不多且会定期过期。