immortalwrt/immortalwrt #1858: Build Error in toolchain-x86_64_gcc-14.2.0_musl

Ardentwheel, 2025-05-06:
### Describe the bug

```
time: toolchain/kernel-headers/compile#0.04#0.02#0.06
make[3]: Entering directory '/home/parallels/immortalwrt/toolchain/musl'
make  -C /home/parallels/immortalwrt/build_dir/toolchain-x86_64_gcc-14.2.0_musl/musl-1.2.5 DESTDIR="/home/parallels/immortalwrt/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/" LIBCC="/home/parallels/immortalwrt/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/lib/gcc/x86_64-openwrt-linux-musl/14.2.0/libgcc.a" all
make[4]: Entering directory '/home/parallels/immortalwrt/build_dir/toolchain-x86_64_gcc-14.2.0_musl/musl-1.2.5'
LC_ALL=C awk -f ./tools/add-cfi.common.awk -f ./tools/add-cfi.x86_64.awk src/ldso/x86_64/tlsdesc.s | x86_64-openwrt-linux-musl-gcc -std=c99 -nostdinc -ffreestanding -fexcess-precision=standard -frounding-math -fno-strict-aliasing -Wa,--noexecstack -D_XOPEN_SOURCE=700 -I./arch/x86_64 -I./arch/generic -Iobj/src/internal -I./src/include -I./src/internal -Iobj/include -I./include  -g -O2 -fno-align-jumps -fno-align-functions -fno-align-loops -fno-align-labels -fira-region=one -fira-hoist-pressure -freorder-blocks-algorithm=simple -fno-prefetch-loop-arrays -fno-tree-ch -pipe -fno-unwind-tables -fno-asynchronous-unwind-tables -ffunction-sections -fdata-sections -Wno-pointer-to-int-cast -Werror=implicit-function-declaration -Werror=implicit-int -Werror=pointer-sign -Werror=pointer-arith -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=discarded-qualifiers -Werror=discarded-array-qualifiers -Waddress -Warray-bounds -Wchar-subscripts -Wduplicate-decl-specifier -Winit-self -Wreturn-type -Wsequence-point -Wstrict-aliasing -Wunused-function -Wunused-label -Wunused-variable -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -x assembler -c -o obj/src/ldso/x86_64/tlsdesc.o -
{standard input}: Assembler messages:
{standard input}:29: Error: bad register expression
make[4]: *** [Makefile:146: obj/src/ldso/x86_64/tlsdesc.o] Error 1
make[4]: Leaving directory '/home/parallels/immortalwrt/build_dir/toolchain-x86_64_gcc-14.2.0_musl/musl-1.2.5'
make[3]: *** [Makefile:32: /home/parallels/immortalwrt/build_dir/toolchain-x86_64_gcc-14.2.0_musl/musl-1.2.5/.built] Error 2
make[3]: Leaving directory '/home/parallels/immortalwrt/toolchain/musl'
time: toolchain/musl/compile#0.11#0.07#0.17
    ERROR: toolchain/musl failed to build.
```

### ImmortalWrt version

974b885c08d3860aa39f857e4451baf1fe47755a

### ImmortalWrt release

974b885c08d3860aa39f857e4451baf1fe47755a

### ImmortalWrt target/subtarget

immortalwrt/toolchain/musl

### Device

x86_64

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

SakuraFallingMad, 2025-05-06:
`ncompatible-pointer-types -Werror=discarded-qualifiers -Werror=discarded-array-qualifiers -Waddress -Warray-bounds -Wchar-subscripts -Wduplicate-decl-specifier -Winit-self -Wreturn-type -Wsequence-point -Wstrict-aliasing -Wunused-function -Wunused-label -Wunused-variable -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -O3 -c -o obj/src/internal/version.o src/internal/version.c                                                                LC_ALL=C awk -f ./tools/add-cfi.common.awk -f ./tools/add-cfi.x86_64.awk src/ldso/x86_64/tlsdesc.s | x86_64-openwrt-linux-musl-gcc -std=c99 -nostdinc -ffreestanding -fexcess-precision=standard -frounding-math -fno-strict-aliasing -Wa,--noexecstack -D_XOPEN_SOURCE=700 -I./arch/x86_64 -I./arch/generic -Iobj/src/internal -I./src/include -I./src/internal -Iobj/include -I./include  -g -O2 -fno-align-jumps -fno-align-functions -fno-align-loops -fno-align-labels -fira-region=one -fira-hoist-pressure -freorder-blocks-algorithm=simple -fno-prefetch-loop-arrays -fno-tree-ch -pipe -fno-unwind-tables -fno-asynchronous-unwind-tables -ffunction-sections -fdata-sections -Wno-pointer-to-int-cast -Werror=implicit-function-declaration -Werror=implicit-int -Werror=pointer-sign -Werror=pointer-arith -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=discarded-qualifiers -Werror=discarded-array-qualifiers -Waddress -Warray-bounds -Wchar-subscripts -Wduplicate-decl-specifier -Winit-self -Wreturn-type -Wsequence-point -Wstrict-aliasing -Wunused-function -Wunused-label -Wunused-variable -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -x assembler -c -o obj/src/ldso/x86_64/tlsdesc.o -
{standard input}: Assembler messages:
{standard input}:29: Error: bad register expression
make[4]: *** [Makefile:146: obj/src/ldso/x86_64/tlsdesc.o] Error 1
make[4]: Leaving directory '/home/compile/immortalwrt/build_dir/toolchain-x86_64_gcc-14.2.0_musl/musl-1.2.5'                                                                        make[3]: *** [Makefile:32: /home/compile/immortalwrt/build_dir/toolchain-x86_64_gcc-14.2.0_musl/musl-1.2.5/.built] Error 2
make[3]: Leaving directory '/home/compile/immortalwrt/toolchain/musl'
time: toolchain/musl/compile#14.12#0.96#14.65
    ERROR: toolchain/musl failed to build.
make[2]: *** [toolchain/Makefile:93`
Same failed results on Debian

SakuraFallingMad, 2025-05-07:
Fixed in https://github.com/openwrt/openwrt/commit/0b1e1943d524afe9a039ef84330a2c1d42819d41, You can look forward to waiting merge official sources.

closed 2025-05-07 by 1715173329