immortalwrt/immortalwrt #1869: tr3000编译master卡壳，编译不成功

localice2021, 2025-05-12:
### Describe the bug

 make[3] -C feeds/luci/modules/luci-compat compile
 make[3] -C package/utils/fitblk compile
 make[3] -C package/base-files compile
 make[3] -C feeds/luci/collections/luci-light compile
 make[3] -C feeds/luci/collections/luci compile
 make[3] -C feeds/luci/applications/luci-app-samba4 compile
 make[3] -C package/kernel/mt76 compile
到这一步就一直不动了，

### ImmortalWrt version

master

### ImmortalWrt release

master

### ImmortalWrt target/subtarget

mediatek_filogic

### Device

tr3000

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-05-12:
说明你网络有问题。

closed 2025-05-12 by 1715173329

localice2021, 2025-05-12:
我是云编译，切换到24.10.1就可以成功