immortalwrt/immortalwrt #1880: 局域网设备互通问题

3846648, 2025-05-19:
### Describe the bug

两台路由器都是ImmortalWrt 24.10.1, 所有设备都在同一网段，没有VLAN，网络拓扑如下：

  终端A（无线） 终端B（有线）						  
        ↓		        ↓
       WiFi		     LAN
        ↓		        ↓
     [360T7 / AP 模式]
             │
            LAN
             ↓
     [RAX3000M 主路由(DHCP、DNS、firewall)]
        ↓		      ↓
       WiFi		    LAN
        ↓		      ↓
  终端C（无线） 终端D（有线）					  

问题现象
终端A与终端D 无法互 ping、终端B与终端C 无法互 ping， 概括来说就是一台路由器的有线客户端和另一台的无线客户端不能互通

经过一个多月的摸索，排查情况如下:
1.重启路由器会短暂正常几天，之后故障会重复出现。
2.终端A 到 终端D，抓包试验，发现arp广播包可以从终端A 到 终端D，但是终端D的arp应答包截止到RAX3000M没有转发给360T7，终端A无法获取终端D的MAC地址而无法通信。
3.后来把360T7刷了原版openwrt，终端B到终端C 可以正常互通了，运行一周没有出现问题，但是终端A与终端D 还是不行。
4.昨天把RAX3000M也刷了原版openwrt，目前运行正常，没有出现上述问题。

### ImmortalWrt version

24.10.1

### ImmortalWrt release

OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.133.85963~b1383cc

### ImmortalWrt target/subtarget

DISTRIB_TARGET='mediatek/filogic'

### Device

RAX3000M、360T7

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

3846648, 2025-05-19:
网络拓扑如下：
[image]

everything411, 2025-05-19:
有相关报告说DSA交换机有时会发送错误的包，如 https://github.com/openwrt/openwrt/issues/18473 。

3846648, 2025-05-19:
后来我在RAX3000M上插了个傻瓜交换机，把设备都插到傻瓜交换机上，AD 可以互通，没有问题

Fitz77777, 2025-06-26:
> 后来我在RAX3000M上插了个傻瓜交换机，把设备都插到傻瓜交换机上，AD 可以互通，没有问题

同样遇到这类问题了，网络拓扑稍微有点不一样，我主副路由直接有一个交换机。我像你这样把本来连在路由的nas放到交换机下之后，能访问nas的页面，但是下载之类的没有速度，很奇怪，不知道后来老哥有没有其他发现 @3846648

3846648, 2025-07-16:
> 1. 卸载dockerd，重启系统。因为原版的dockerd会打开bridge-call-iptables
> 2. wifi如果打开了隔离客户端也要关掉
> 3. 如果wifi路由器有ap模式，别用，而是直接桥接到lan口，关闭dhcp

您说的这些场景都不存在，现在刷的原版openwrt，同样的配置两个多月了没出现这样的问题。

skbeh, 2025-08-10:
https://github.com/nbd168/bridger/issues/6