immortalwrt/immortalwrt #1905: rk3588 USB3.0 无法识别

3wlh, 2025-06-03:
### Describe the bug

rk3588 USB3.0无法识别插入设备，无识别，数据灯不亮，接口无供电

### ImmortalWrt version

24.10.1

### ImmortalWrt release

ImmortalWrt 24.10.1 r33048-cc720ea55a71 / LuCI openwrt-24.10 branch 25.120.37759~99d252e

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

friendlyarm nanopi r6s，xunlong orangepi 5 plus

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-06-04:
No log no solution.

3wlh, 2025-06-04:
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.86 (buildbot@buildbot.infra.immortalwrt.org) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r33048-cc720ea55a71) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Apr 15 11:38:42 2025
[    0.000000] KASLR enabled
[    0.000000] Machine model: FriendlyElec NanoPi R6S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000feb50000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000002ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000]   node   0: [mem 0x00000002f0000000-0x00000002ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000002ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x00000000ef000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s78696 r8192 d31896 u118784
[    0.000000] pcpu-alloc: s78696 r8192 d31896 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=2M console=tty1 console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfeb50000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2063880
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 8.
[    0.000000] software IO TLB: mapped [mem 0x00000000e8b98000-0x00000000ecb98000] (64MB)
[    0.000000] Memory: 8117072K/8386560K available (10816K kernel code, 3086K rwdata, 7632K rodata, 3072K init, 760K bss, 253104K reserved, 16384K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=8.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=8
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 480 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [424:479]
[    0.000000] GICv3: Using MBI frame 0x00000000fe610000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fe680000
[    0.000000] ITS [mem 0xfe640000-0xfe65ffff]
[    0.000000] GIC: enabling workaround for ITS: Rockchip erratum RK3588001
[    0.000000] ITS@0x00000000fe640000: allocated 8192 Devices @1000e0000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fe640000: allocated 32768 Interrupt Collections @1000f0000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] ITS [mem 0xfe660000-0xfe67ffff]
[    0.000000] GIC: enabling workaround for ITS: Rockchip erratum RK3588001
[    0.000000] ITS@0x00000000fe660000: allocated 8192 Devices @100100000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fe660000: allocated 32768 Interrupt Collections @100110000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x0000000100120000
[    0.000000] GIC: using cache flushing for LPI property table
[... 449 lines cut ...]
[   15.884155] dwmac4: Master AXI performs any burst length
[   15.884651] rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
[   15.885322] rk_gmac-dwmac fe1c0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   15.886134] rk_gmac-dwmac fe1c0000.ethernet eth0: registered PTP clock
[   15.886824] rk_gmac-dwmac fe1c0000.ethernet eth0: FPE workqueue start
[   15.887400] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[... cut ...]
[  531.105011] kmodloader: done loading kernel modules from /etc/modules.d/*
[  531.134339] kmodloader: loading kernel modules from /etc/modules.d/*
[  531.138318] kmodloader: done loading kernel modules from /etc/modules.d/*
[  531.170505] kmodloader: loading kernel modules from /etc/modules.d/*
[  531.175075] kmodloader: done loading kernel modules from /etc/modules.d/*
[  531.205234] kmodloader: loading kernel modules from /etc/modules.d/*
[  531.210577] kmodloader: done loading kernel modules from /etc/modules.d/*

3wlh, 2025-06-04:
要系统日志吗还是内核日志

3wlh, 2025-06-06:
一样是 USB3.0 无法识别

3wlh, 2025-06-14:
NanoPi R6S USB3.0 首次运行能识别，重启后无法识别，sub3.0无供电

ibrah3m, 2025-08-27:
I have same problem no power and doesn't recognize external interfaces

ibrah3m, 2025-08-28:
@1715173329 is the USB 3.0 controllers. disabled in the kernel ?

dutian312, 2025-10-22:
我用的zero3 也是无法识别13pin扩展的usb  我改了一下dtb文件就正常了  刚开始也是不供电无法发现设备，参考官方提供的镜像调整了一下dtb文件就好了 你试试

adinxiang, 2025-12-09:
已经使用到了24.10.4版本的系统，orange pi5 plus硬件，usb3依旧无法识别。