immortalwrt/immortalwrt #1906: master分支最新，接口存在严重问题导致没网

shjy223534, 2025-06-03:
### Describe the bug

接口处设置好lan和wan后，接口方面各种信息都是显示完整的包括拨号，但是概览方面的网络会出现如下图的问题，导致即使接口处即使任何信息都显示正常，但是是没有网络的

### ImmortalWrt version

r34409-57f8e83584

### ImmortalWrt release

master

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Radxa E25 Carrier Board

### Image kind

Official downloaded image

### Steps to reproduce

编译好master版本的6.12.31版本后，观察概览的网络会有如图所示的错误

[image]，而且cpu占用也会重新出现问题，出现如图所示的持续高占用，编译的时候autocor是选择了的，但还是会出现持续高占用

[image]

### Actual behaviour

不仅占用重新出现高占用的问题，概览里面的网络也出现了错误，这个错误导致没有网，接口已经正确配置了lan和wan,而且lan可以正确分配ip，内网可以访问，wan口也正常拨号成功，但是就是没有网

### Expected behaviour

无

### Additional info

错误日志

[新建 文本文档.txt](https://github.com/user-attachments/files/20581758/default.txt)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

shjy223534, 2025-06-03:
还存在一个编译上的问题，那就是第三方的tubroaac ,正常下载到源码里面后，选择luci-app-tubroacc后无法正常保存设置，6.12.30内核是正常保存的

libriunk, 2025-06-04:
> > 还存在一个编译上的问题，那就是第三方的tubroaac ，正常下载到源码里面后，选择luci-app-tubroacc后无法正常保存设置，6.12.30内核是正常保存的
> 
> 测试了官方6.12.31是可以正常保存设置的 ，imm的会不断提示正在写入配置，不管保存多少次都是这个提示，只要取消选择tubroaac才可以正常保存配置

没有必要引入turboacc插件，您路由器没有 **硬件流量卸载** 路由器不做落地机也没必要开bbr