immortalwrt/immortalwrt #1909: A-040W-Q刷入24.10.1后mac地址不对

DuDuF-1019, 2025-06-05:
看下载地址和文件A-040W-Q用的是MSG1500同一个系统，刷入后MAC地址不对。是需要刷入MSG1500的UBOOT吗？我现在用的breed。但是mtd_write写不进去，超过了256kb。需要怎么操作才能正常使用，感谢。目前刷的是Padavan。
dev:    size   erasesize  name
mtd0: 00040000 00020000 "Bootloader"
mtd1: 00040000 00020000 "BootEnv"
mtd2: 00080000 00020000 "Config"
mtd3: 00040000 00020000 "Factory"
mtd4: 00160000 00020000 "Kernel"
mtd5: 016a0000 00020000 "RootFS"
mtd6: 00400000 00020000 "Storage"
mtd7: 06240000 00020000 "RWFS"
mtd8: 01800000 00020000 "Firmware_Stub"
mtd9: 07f80000 00020000 "ALL"

DuDuF-1019, 2025-06-05:
感谢，但是还是遇到了问题。
Permission denied，这个怎么解决？麻烦了您了。

1715173329, 2025-06-05:
code 右上角有个复制按钮，直接粘贴进命令行里面就行了

1715173329, 2025-06-05:
dd if=/dev/mtd1  of=/tmp/mtd1
利用 scp 等工具下载 /tmp/mtd1 然后传上来看看