immortalwrt/immortalwrt #1935: BananaPi R3 mini无法刷入NAND

LZY-SPCA, 2025-06-19:
### Describe the bug

[image]
我参考这张图中的内容，在emmc系统中尝试刷入NAND，1234条执行都没有问题，刷入的bl2和fip均来自于firmware selector，但在第五条时，我使用Load recovery system via TFTP then write to NAND，系统启动后卡住，重启报错

[image]

### ImmortalWrt version

24.10.1

### ImmortalWrt release

24.10.1

### ImmortalWrt target/subtarget

24.10.1

### Device

BananaPi-R3-mini

### Image kind

Official downloaded image

### Additional info

我尝试在NAND下通过tftp启动系统，系统可以启动，但是mtd分区表只有两项

[image]

但在emmc系统下刷机的时候mtd是完整的

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-06-19:
这是过时教程，具体自行参考 openwrt 官方文档。

closed 2025-06-19 by 1715173329

LZY-SPCA, 2025-06-19:
请问方便提供一下可用的刷机教程吗？非常感谢

libiunc, 2025-07-13:
> 请问方便提供一下可用的刷机教程吗？非常感谢

https://github.com/immortalwrt/immortalwrt/commit/b03d3644cfe9eb8db2a3441c55c84b076ba7cd08

xlighting2017, 2025-07-13:
> > 请问方便提供一下可用的刷机教程吗？非常感谢
> 
> [b03d364](https://github.com/immortalwrt/immortalwrt/commit/b03d3644cfe9eb8db2a3441c55c84b076ba7cd08) 好吧就是这个

这个 commit msg 里面，NAND 部分有几处错误，需要按下面的 粗体部分 修正：

ubiupdatevol /dev/ubi0_0 /tmp/openwrt-*-bananapi_bpi-r3-mini-snand-bl31-uboot.fip
    cd /lib/firmware/airoha
    cat EthMD32.dm.bin EthMD32.DSP.bin > /tmp/en8811h-fw.bin
    ubimkvol /dev/ubi0 -N **en8811h-fw** -n 1 -s 147456 -t static
    ubiupdatevol /dev/ubi0_1 /tmp/en8811h-fw.bin
    ubimkvol /dev/ubi0 -n 2 -N ubootenv -s 126976
    ubimkvol /dev/ubi0 -n 3 -N ubootenv2 -s 126976
    volsize=$(wc -c < /tmp/openwrt-*-bananapi_bpi-r3-mini-initramfs-recovery.itb)
    ubimkvol /dev/ubi0 -n 4 -N recovery -s $volsize
    ubiupdatevol /dev/ubi0_4 /tmp/openwrt-*-bananapi_bpi-r3-mini-initramfs-recovery.itb
    volsize=$(wc -c < /tmp/openwrt-*-bananapi_bpi-r3-mini-squashfs-sysupgrade.itb)
    ubimkvol /dev/ubi0 -n **5 -N fit** -s $volsize
    ubiupdatevol **/dev/ubi0_5** /tmp/openwrt-*-bananapi_bpi-r3-mini-squashfs-sysupgrade.itb

或者按照这个文档：   https://openwrt.org/toh/sinovoip/bananapi_bpi_r3_mini#installation ， 特别注意最下方 p.s./ p.p.s. 的部分

LZY-SPCA, 2025-07-26:
需要注意的是必须要首先将emmc刷到24.10版本，否则在emmc系统中看到的是老的NAND分区也就没有办法刷NAND