immortalwrt/immortalwrt #1967: immortalwrt 不支持 8168 网卡

qq122356, 2025-07-11:
### Describe the bug

联想扬天T4900-v上（单网卡）安装immortalwrt 24.10.2后，发现网络不通，偶尔会联通，但是断断续续，
然后加装一块网卡（博科的网卡），安装后通讯都正常
在另一台主机上同样安装后，同样是网络不通，偶尔会联通，但是断断续续
不通的网卡型号是Realtek r8168
所有操作用的网线网口都是同一个

### ImmortalWrt version

24.10.2

### ImmortalWrt release

ImmortalWrt 24.10.2 r33247-467867283bb9

### ImmortalWrt target/subtarget

ImmortalWrt 24.10.2 r33247-467867283bb9

### Device

联想扬天T4900-v

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-07-11:
卸载掉 kmod-r8168 安装 kmod-r8169 即可

qq122356, 2025-07-14:
已找到：
https://downloads.openwrt.org/releases/24.10.2/targets/x86/64/kmods/6.6.93-1-1745ebad77278f5cdc8330d17a3f43d6/kmod-r8169_6.6.93-r1_x86_64.ipk

closed 2025-07-15 by qq122356