immortalwrt/immortalwrt #1981: R2S不定期发生断连

wm1128, 2025-07-26:
### Describe the bug

R2S使用v24.10.2版本固件不定期发生断连，偶尔会自动恢复，偶尔只能通过强制重启才能恢复。
通过查询日志发现有内核报错
rk_gmac-dwmac ff540000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 0 timed out 5276 ms
rk_gmac-dwmac ff540000.ethernet eth0: Reset adapter.
24.10.2是我使用的第一个Immortalwrt版本，不确定之前的版本是否有类似问题。
问题描述和这个issue里提到的现象很类似，可供参考，https://github.com/openwrt/openwrt/issues/17948

### ImmortalWrt version

v24.10.2

### ImmortalWrt release

 24.10.2 (r33247-467867283bb9)

### ImmortalWrt target/subtarget

 rockchip/armv8

### Device

 FriendlyARM NanoPi R2S

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-08-25:
> 我也遇到了同样的问题，但我是x86，rtl8168

可以尝试安装 kmod-r8169 并卸载掉 kmod-r8168，之后重启。

halfu, 2026-03-22:
我做了两件事情：
1、更换驱动：
`# 更新软件源
opkg update

# 卸载 Linux 内核原生的 RTL8152 驱动
opkg remove kmod-usb-net-rtl8152

# 安装基于 Realtek 官方源码移植、深度优化的 Vendor 驱动
opkg install kmod-usb-net-rtl8152-vendor

# 重启设备以加载新驱动
reboot
`

2、关闭接口的硬件流控 (Flow Control) 
`ethtool -A eth0 rx off tx off
ethtool -A eth1 rx off tx off
`

然后使用iperf3进行极限测试，正反各测试了60秒，未出现网卡卸载的问题。

closed 2026-09-15 by 1715173329