immortalwrt/immortalwrt #1983: gettext-full编译错误

shjy223534, 2025-07-27:
### Describe the bug

编译存在报错：make[8]： Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/misc'
make[8]： 'all' 无事可做。
make[8]：离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/misc'
使所有目录都在 po
make[8]：进入目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/po'
Makefile：596：警告：覆盖目标“[en@quot.po-create](mailto:en@quot.po-create)”
的配方 Rules-quot：16：警告：忽略目标“[en@quot.po-create](mailto:en@quot.po-create)”
的旧配方 Makefile：598：警告：覆盖配方对于目标 '[en@boldquot.po-create](mailto:en@boldquot.po-create)'
Rules-quot：18： 警告：忽略目标 '[en@boldquot.po-create](mailto:en@boldquot.po-create)'
的旧配方 Makefile：604： 警告：覆盖目标 '.insert-header.po-update-en'
的配方 Rules-quot：24： 警告：忽略目标 '.insert-header.po-update-en'
的旧配方 Makefile：639： 警告：覆盖目标 '[en@quot.insert-header](mailto:en@quot.insert-header)'
的配方Rules-quot：59：警告：忽略目标“[en@quot.insert-header](mailto:en@quot.insert-header)”
的旧配方 Makefile：642：警告：覆盖目标“[en@boldquot.insert-header](mailto:en@boldquot.insert-header)”
的配方 Rules-quot：62：警告：忽略目标“[en@boldquot.insert-header](mailto:en@boldquot.insert-header)”
的旧配方 Makefile：646：警告：覆盖目标“mostlyclean-quot”
的配方 Rules-quot：66：警告：忽略目标“mostlyclean-quot”
的旧配方make[8]：对“所有人”无事可做。
make[8]：离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/po'
使所有项目
make[8]：进入目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/projects'
make[8]：对 'all' 无事可做。
make[8]：离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/projects'
将所有内容都放在 src 中
make[8]：进入目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make all-am
make[9]：进入目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
/home/sh/imm/staging_dir/host/bin/gcc -DLOCALEDIR=“/home/sh/imm/staging_dir/hostpkg/share/locale” -DUSEJAVA=0 -DGETTEXTJAR=“/home/sh/imm/staging_dir/hostpkg/share/gettext”“/gettext.jar” -DBINDIR=“/home/sh/imm/staging_dir/hostpkg/bin” -DLIBDIR=“/home/sh/imm/staging_dir/hostpkg/lib” -DLIBEXECDIR=“/home/sh/imm/staging_dir/hostpkg/libexec” -DGETTEXTDATADIR=“/home/sh/imm/staging_dir/hostpkg/share/gettext” -DPROJECTSDIR=“/home/sh/imm/staging_dir/hostpkg/share/gettext“”/projects“ -DEXEEXT=”“ -DHAVE_CONFIG_H -I. -I..-我。。-我。。/tree-sitter-0.23.2/lib/include -I.。/libgrep -I.。/libgrep -I.。/gnulib-lib -I.。/gnulib-lib -I.。/../gettext-runtime/intl -I../../gettext-runtime/intl -DINSTALLDIR=“/home/sh/imm/staging_dir/hostpkg/bin” -I/home/sh/imm/staging_dir/host/include -I/home/sh/imm/staging_dir/hostpkg/include -I/home/sh/imm/staging_dir/target-aarch64_generic_musl/host/include -O2 -I/home/sh/imm/staging_dir/host/include -I/home/sh/imm/staging_dir/hostpkg/include -I/home/sh/imm/staging_dir/target-aarch64_generic_musl/host/include -DPIC -fPIC -c -o msgcmp-msgcmp.o msgcmp.c
在文件中包含msgcmp.c：35：
msgcmp.c：在函数“main”中：
msgcmp.c：110：46：错误：“BISON_LOCALEDIR”未声明（在此函数中首次使用）
110 |bindtextdomain （“bison-runtime”， relocate （BISON_LOCALEDIR））;
|^~~~~~~~~~~~~~~
../gnulib-lib/relocatable.h：114：29：注意：在宏“重新定位”
的定义中 114 |#define relocate（路径名） （路径名）
|^~~~~~~~
msgcmp.c：110：46：注意：每个未声明的标识符仅针对它出现在 110 |
bindtextdomain （“bison-runtime”， relocate （BISON_LOCALEDIR））;
|^~~~~~~~~~~~~~~
../gnulib-lib/relocatable.h：114：29：注意：在宏“重新定位”
的定义中 114 |#define relocate（路径名） （路径名）
|^~~~~~~~
make[9]： *** [Makefile：5539： msgcmp-msgcmp.o] 错误 1
make[9]： 离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make[8]： *** [Makefile：4374： all] 错误 2
make[8]： 离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make[7]： *** [Makefile：3591： all-recursive] 错误 1
make[7]： 离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools'
make[6]： *** [Makefile：3434： all] 错误 2
make[6]： 离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools'
make[5]： *** [Makefile：411： all-recursive] 错误 1
make[5]： 离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1'
make[4]： *** [Makefile：367： all] 错误 2
make[4]：离开目录 '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1'
make[3]： *** [Makefile：188： /home/sh/imm/build_dir/hostpkg/gettext-0.24.1/.built] 错误 2
make[3]：离开目录 '/home/sh/imm/package/libs/gettext-full'
时间：package/libs/gettext-full/host-compile#0.23#0.28#0.48
错误：package/libs/gettext-full [host] 构建失败。
make[2]： *** [package/Makefile：181： package/libs/gettext-full/host/compile] 错误 1
make[2]： 离开目录 '/home/sh/imm'
make[1]： *** [package/Makefile：177： /home/sh/imm/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] 错误 2
make[1]： 离开目录 '/home/sh/imm'
make： *** [/home/sh/imm/include/toplevel.mk：233：world] 错误 2test -f 'msgcmp.c' || echo './'

### ImmortalWrt version

/

### ImmortalWrt release

/

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

Radxa E25 Carrier Board

### Image kind

Official downloaded image

### Additional info

/

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

shjy223534, 2025-07-28:
> 3\. rror 2

Making all in projects
make[8]: Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/projects'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/projects'
Making all in src
make[8]: Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make  all-am
make[9]: Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
/home/sh/imm/staging_dir/host/bin/gcc -DLOCALEDIR=\"/home/sh/imm/staging_dir/hostpkg/share/locale\" -DUSEJAVA=0 -DGETTEXTJAR=\"/home/sh/imm/staging_dir/hostpkg/share/gettext\"\"/gettext.jar\" -DBINDIR=\"/home/sh/imm/staging_dir/hostpkg/bin\" -DLIBDIR=\"/home/sh/imm/staging_dir/hostpkg/lib\" -DLIBEXECDIR=\"/home/sh/imm/staging_dir/hostpkg/libexec\" -DGETTEXTDATADIR=\"/home/sh/imm/staging_dir/hostpkg/share/gettext\" -DPROJECTSDIR=\"/home/sh/imm/staging_dir/hostpkg/share/gettext\"\"/projects\" -DEXEEXT=\"\" -DHAVE_CONFIG_H -I. -I..  -I.. -I../tree-sitter-0.23.2/lib/include -I../libgrep -I../libgrep -I../gnulib-lib -I../gnulib-lib -I../../gettext-runtime/intl -I../../gettext-runtime/intl -DINSTALLDIR=\"/home/sh/imm/staging_dir/hostpkg/bin\" -I/home/sh/imm/staging_dir/host/include -I/home/sh/imm/staging_dir/hostpkg/include -I/home/sh/imm/staging_dir/target-aarch64_generic_musl/host/include  -O2 -I/home/sh/imm/staging_dir/host/include -I/home/sh/imm/staging_dir/hostpkg/include -I/home/sh/imm/staging_dir/target-aarch64_generic_musl/host/include -DPIC -fPIC -c -o msgcmp-msgcmp.o `test -f 'msgcmp.c' || echo './'`msgcmp.c
In file included from msgcmp.c:35:
msgcmp.c: In function 'main':
msgcmp.c:110:46: error: 'BISON_LOCALEDIR' undeclared (first use in this function)
  110 |   bindtextdomain ("bison-runtime", relocate (BISON_LOCALEDIR));
      |                                              ^~~~~~~~~~~~~~~
../gnulib-lib/relocatable.h:114:29: note: in definition of macro 'relocate'
  114 | #define relocate(pathname) (pathname)
      |                             ^~~~~~~~
msgcmp.c:110:46: note: each undeclared identifier is reported only once for each function it appears in
  110 |   bindtextdomain ("bison-runtime", relocate (BISON_LOCALEDIR));
      |                                              ^~~~~~~~~~~~~~~
../gnulib-lib/relocatable.h:114:29: note: in definition of macro 'relocate'
  114 | #define relocate(pathname) (pathname)
      |                             ^~~~~~~~
make[9]: *** [Makefile:5539: msgcmp-msgcmp.o] Error 1
make[9]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make[8]: *** [Makefile:4374: all] Error 2
make[8]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make[7]: *** [Makefile:3591: all-recursive] Error 1
make[7]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools'
make[6]: *** [Makefile:3434: all] Error 2
make[6]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools'
make[5]: *** [Makefile:411: all-recursive] Error 1
make[5]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1'
make[4]: *** [Makefile:367: all] Error 2
make[4]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1'
make[3]: *** [Makefile:188: /home/sh/imm/build_dir/hostpkg/gettext-0.24.1/.built] Error 2
make[3]: Leaving directory '/home/sh/imm/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#0.19#0.32#0.48
    ERROR: package/libs/gettext-full [host] failed to build.
make[2]: *** [package/Makefile:181: package/libs/gettext-full/host/compile] Error 1
make[2]: Leaving directory '/home/sh/imm'
make[1]: *** [package/Makefile:177: /home/sh/imm/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sh/imm'
make: *** [/home/sh/imm/include/toplevel.mk:233：world] 错误 2

shjy223534, 2025-07-28:
> 1. diffconfig呢？
> 2. 是j1跑的吗？
> 3. 错误日志里面搜“rror 2”来排除是哪个玩意出问题了
> 
> 我上次遇到类似的问题是kmod-32炸了

Making all in projects
make[8]: Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/projects'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/projects'
Making all in src
make[8]: Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make  all-am
make[9]: Entering directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
/home/sh/imm/staging_dir/host/bin/gcc -DLOCALEDIR=\"/home/sh/imm/staging_dir/hostpkg/share/locale\" -DUSEJAVA=0 -DGETTEXTJAR=\"/home/sh/imm/staging_dir/hostpkg/share/gettext\"\"/gettext.jar\" -DBINDIR=\"/home/sh/imm/staging_dir/hostpkg/bin\" -DLIBDIR=\"/home/sh/imm/staging_dir/hostpkg/lib\" -DLIBEXECDIR=\"/home/sh/imm/staging_dir/hostpkg/libexec\" -DGETTEXTDATADIR=\"/home/sh/imm/staging_dir/hostpkg/share/gettext\" -DPROJECTSDIR=\"/home/sh/imm/staging_dir/hostpkg/share/gettext\"\"/projects\" -DEXEEXT=\"\" -DHAVE_CONFIG_H -I. -I..  -I.. -I../tree-sitter-0.23.2/lib/include -I../libgrep -I../libgrep -I../gnulib-lib -I../gnulib-lib -I../../gettext-runtime/intl -I../../gettext-runtime/intl -DINSTALLDIR=\"/home/sh/imm/staging_dir/hostpkg/bin\" -I/home/sh/imm/staging_dir/host/include -I/home/sh/imm/staging_dir/hostpkg/include -I/home/sh/imm/staging_dir/target-aarch64_generic_musl/host/include  -O2 -I/home/sh/imm/staging_dir/host/include -I/home/sh/imm/staging_dir/hostpkg/include -I/home/sh/imm/staging_dir/target-aarch64_generic_musl/host/include -DPIC -fPIC -c -o msgcmp-msgcmp.o `test -f 'msgcmp.c' || echo './'`msgcmp.c
In file included from msgcmp.c:35:
msgcmp.c: In function 'main':
msgcmp.c:110:46: error: 'BISON_LOCALEDIR' undeclared (first use in this function)
  110 |   bindtextdomain ("bison-runtime", relocate (BISON_LOCALEDIR));
      |                                              ^~~~~~~~~~~~~~~
../gnulib-lib/relocatable.h:114:29: note: in definition of macro 'relocate'
  114 | #define relocate(pathname) (pathname)
      |                             ^~~~~~~~
msgcmp.c:110:46: note: each undeclared identifier is reported only once for each function it appears in
  110 |   bindtextdomain ("bison-runtime", relocate (BISON_LOCALEDIR));
      |                                              ^~~~~~~~~~~~~~~
../gnulib-lib/relocatable.h:114:29: note: in definition of macro 'relocate'
  114 | #define relocate(pathname) (pathname)
      |                             ^~~~~~~~
make[9]: *** [Makefile:5539: msgcmp-msgcmp.o] Error 1
make[9]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make[8]: *** [Makefile:4374: all] Error 2
make[8]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools/src'
make[7]: *** [Makefile:3591: all-recursive] Error 1
make[7]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools'
make[6]: *** [Makefile:3434: all] Error 2
make[6]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1/gettext-tools'
make[5]: *** [Makefile:411: all-recursive] Error 1
make[5]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1'
make[4]: *** [Makefile:367: all] Error 2
make[4]: Leaving directory '/home/sh/imm/build_dir/hostpkg/gettext-0.24.1'
make[3]: *** [Makefile:188: /home/sh/imm/build_dir/hostpkg/gettext-0.24.1/.built] Error 2
make[3]: Leaving directory '/home/sh/imm/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#0.19#0.32#0.48
    ERROR: package/libs/gettext-full [host] failed to build.
make[2]: *** [package/Makefile:181: package/libs/gettext-full/host/compile] Error 1
make[2]: Leaving directory '/home/sh/imm'
make[1]: *** [package/Makefile:177: /home/sh/imm/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sh/imm'
make: *** [/home/sh/imm/include/toplevel.mk:233：world] 错误 2

shjy223534, 2025-07-28:
> 1. diffconfig呢？
> 2. 是j1跑的吗？
> 3. 错误日志里面搜“rror 2”来排除是哪个玩意出问题了
> 
> 我上次遇到类似的问题是kmod-32炸了

可以帮忙看看嘛，现在一直卡在这个地方

airjinkela, 2025-07-28:
看一下gcc 版本
```BASH
gcc -v

g++ -v
```

shjy223534, 2025-07-28:
> g++ -v

这两个都是11.4.0

airjinkela, 2025-07-28:
https://github.com/openwrt/openwrt/pull/16522#issuecomment-3124684791

应该是上游炸了

shjy223534, 2025-07-28:
> [openwrt/openwrt#16522（评论）](https://github.com/openwrt/openwrt/pull/16522#issuecomment-3124684791)
> 
> 应该是上游炸了

这上游怎么老炸

1715173329, 2025-07-28:
开发分支不炸等着进 stable 炸吗？解决不了/想要 stable 就去用 stable 分支