immortalwrt/immortalwrt #1985: X86最新Master分支编译失败

jh0072110, 2025-07-29:
### Describe the bug

如日志所示，package/libs/gettext-full插件出现了问题

### ImmortalWrt version

https://github.com/immortalwrt/immortalwrt/commit/14fc614405e045a4ec05fb32093e806112e92d93

### ImmortalWrt release

/

### ImmortalWrt target/subtarget

/

### Device

X86

### Image kind

Self-built image

### Additional info

[0_build.txt](https://github.com/user-attachments/files/21488082/0_build.txt)
以上部分为编译日志，以下内容为chatgpt对该日志的分析（我瞎问的不一定准确希望能起到帮助，起不到帮助天灵大佬就直接无视就行）：

从编译日志看，真正的编译错误出现在 package/libs/gettext-full 的宿主工具链（host-compile）阶段：

在构建 gettext-tools/src/msgcmp.c 时，编译器报错未定义宏 BISON_LOCALEDIR：

msgcmp.c: In function 'main':  
msgcmp.c:110:46: error: 'BISON_LOCALEDIR' undeclared (first use in this function)  
  110 |   bindtextdomain ("bison-runtime", relocate (BISON_LOCALEDIR));  

这说明在 msgcmp.c 第 110 行使用了未定义的符号 BISON_LOCALEDIR，导致生成 msgcmp-msgcmp.o 失败 
.

随后所有递归的 make all 调用都因该错误终止：

/workdir/openwrt/build_dir/hostpkg/gettext-0.24.1/gettext-tools 模块编译失败，返回 Error 1 

导致整个 package/libs/gettext-full [host] 构建失败，Makefile 报 ERROR: package/libs/gettext-full [host] failed to build. 

最终顶层目标 world 编译中断，退出码 2 
.

结论：

失败点在 package/libs/gettext-full 的 host-compile 阶段，具体错误是 msgcmp.c 中使用了未声明的宏 BISON_LOCALEDIR。

需要检查 gettext-full 的 Makefile 或补丁，确保在编译宿主工具时为 msgcmp.c 提供正确的 BISON_LOCALEDIR 定义（通常来自 bison-runtime 的安装路径宏）。

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

guhill1, 2025-07-30:
github action x86 编译同样出错

jh0072110, 2025-07-30:
看到上游已经有该问题的issue以及修复方式：
https://github.com/openwrt/openwrt/issues/19601#issuecomment-3135275860
期待天灵大佬能做一个fix修复

smallprogram, 2025-07-30:
等着合并然后同步把

closed 2025-07-30 by 1715173329