immortalwrt/immortalwrt #1990: Replace broken pcat-manager

rbqvq, 2025-08-01:
I'm just rewritten a tiny photonicat manager [pcat-mgr](https://gitlab.com/CoiaPrant/pcat-mgr) to replace broken pcat-manager

- Not enable watchdog, I think if program panic will let user device in a reboot loop :(
- Added a battery FUSE mount to `/sys/class/powe_supply` with https://github.com/openwrt/luci/pull/7881

1715173329, 2025-08-31:
done with immortalwrt/packages@623e50e6a020

rbqvq, 2025-10-25:
晚点我给内置表更新一下吧
他们硬编码了一个表然后又用配置文件更新表
感觉有点**

szescxz, 2025-10-25:
I think they might have a good reason to do so.

According to [the official wiki](https://photonicat.com/wiki/Photonicat_%E8%87%AA%E5%8A%A8%E5%85%B3%E6%9C%BA%E7%94%B5%E5%8E%8B%E4%BF%AE%E6%94%B9), users who installed the RM500U module requires a minimum voltage of 3.6V, while other users may go as low as 3.4V. This creates a situation where users with different cellular modules might need different tables (at least for the discharge table).

rbqvq, 2025-10-25:
Add a table for RM500U (