immortalwrt/immortalwrt #2006: ImmortalWrt uhttpd正常启动后不定时的去世

xcysy32, 2025-08-12:
### Describe the bug

root@ImmortalWrt:~# logread | grep uhttpd
Mon Aug 11 23:51:38 2025 kern.info kernel: [   36.312707] uhttpd[2769]: segfault at 7fbf56db8680 ip 00007fbf5702b4d0 sp 00007ffd89578048 error 4 in libcrypto.so.3[7fbf56fed000+26a000] likely on CPU 3 (core 3, socket 0)
Mon Aug 11 23:52:46 2025 daemon.err uhttpd[5495]: [info] luci: accepted login on / for root from 240e:b8f:5d1:be00:4c7c:f3:da41:8e8
Mon Aug 11 23:53:16 2025 kern.info kernel: [  133.999871] uhttpd[5495]: segfault at 7f37627718c0 ip 00007f376298d4d0 sp 00007ffe87ea01d8 error 4 in libcrypto.so.3[7f376294f000+26a000] likely on CPU 3 (core 3, socket 0)
Mon Aug 11 23:53:47 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:47 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Mon Aug 11 23:53:47 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:47 ERROR Defaulting to storing config in current directory.
Mon Aug 11 23:53:47 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:47 ERROR Use --config flag to workaround.
Mon Aug 11 23:53:47 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:47 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Mon Aug 11 23:53:47 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:47 ERROR Defaulting to storing config in current directory.
Mon Aug 11 23:53:47 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:47 ERROR Use --config flag to workaround.
Mon Aug 11 23:53:50 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:50 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Mon Aug 11 23:53:50 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:50 ERROR Defaulting to storing config in current directory.
Mon Aug 11 23:53:50 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:50 ERROR Use --config flag to workaround.
Mon Aug 11 23:53:50 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:50 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Mon Aug 11 23:53:50 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:50 ERROR Defaulting to storing config in current directory.
Mon Aug 11 23:53:50 2025 daemon.err uhttpd[8563]: 2025/08/11 15:53:50 ERROR Use --config flag to workaround.
Mon Aug 11 23:57:39 2025 kern.info kernel: [  397.306363] uhttpd[8563]: segfault at 7f777497f050 ip 00007f7774bae4d0 sp 00007ffd5e661f38 error 4 in libcrypto.so.3[7f7774b70000+26a000] likely on CPU 3 (core 3, socket 0)
Mon Aug 11 23:57:46 2025 kern.info kernel: [  404.041639] uhttpd[10949]: segfault at 7f99fed440e0 ip 00007f99fef894d0 sp 00007ffff7a90388 error 4 in libcrypto.so.3[7f99fef4b000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 00:03:47 2025 kern.info kernel: [  765.593820] uhttpd[10950]: segfault at 7f4a00524e60 ip 00007f4a008404d0 sp 00007ffd365311a8 error 4 in libcrypto.so.3[7f4a00802000+26a000] likely on CPU 0 (core 0, socket 0)
Tue Aug 12 00:04:41 2025 kern.info kernel: [  819.434423] uhttpd[11728]: segfault at 7f74ffe4b050 ip 00007f75001264d0 sp 00007ffef1608af8 error 4 in libcrypto.so.3[7f75000e8000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 00:04:41 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 49 seconds since last crash
Tue Aug 12 10:10:28 2025 daemon.err uhttpd[28428]: [info] luci: accepted login on / for root from 240e:b8f:5d1:be00:868e:15a8:7843:3766
Tue Aug 12 10:13:12 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:12 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Tue Aug 12 10:13:12 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:12 ERROR Defaulting to storing config in current directory.
Tue Aug 12 10:13:12 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:12 ERROR Use --config flag to workaround.
Tue Aug 12 10:13:12 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:12 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Tue Aug 12 10:13:12 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:12 ERROR Defaulting to storing config in current directory.
Tue Aug 12 10:13:12 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:12 ERROR Use --config flag to workaround.
Tue Aug 12 10:13:15 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:15 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Tue Aug 12 10:13:15 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:15 ERROR Defaulting to storing config in current directory.
Tue Aug 12 10:13:15 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:15 ERROR Use --config flag to workaround.
Tue Aug 12 10:13:15 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:15 ERROR Couldn't find configuration directory nor home directory: exec: "getent": executable file not found in $PATH
Tue Aug 12 10:13:15 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:15 ERROR Defaulting to storing config in current directory.
Tue Aug 12 10:13:15 2025 daemon.err uhttpd[28428]: 2025/08/12 02:13:15 ERROR Use --config flag to workaround.
Tue Aug 12 10:14:01 2025 kern.info kernel: [37380.259404] uhttpd[28428]: segfault at 7fcb03db9e60 ip 00007fcb03fd94d0 sp 00007ffdb188ca08 error 4 in libcrypto.so.3[7fcb03f9b000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 12 10:22:27 2025 daemon.err uhttpd[29976]: [info] luci: accepted login on / for root from 240e:b8f:5d1:be00:868e:15a8:7843:3766
Tue Aug 12 10:27:07 2025 kern.info kernel: [38166.035765] uhttpd[29976]: segfault at 7f98cee95680 ip 00007f98cf0c44d0 sp 00007ffe4a1eeaa8 error 4 in libcrypto.so.3[7f98cf086000+26a000] likely on CPU 0 (core 0, socket 0)
Tue Aug 12 10:33:09 2025 kern.info kernel: [38527.565232] uhttpd[4166]: segfault at 7facc2a39dd0 ip 00007facc2c684d0 sp 00007ffea7ce2978 error 4 in libcrypto.so.3[7facc2c2a000+26a000] likely on CPU 1 (core 1, socket 0)
Tue Aug 12 10:34:57 2025 kern.info kernel: [38635.981862] uhttpd[8306]: segfault at 7fa6bb0c5ef0 ip 00007fa6bb2d24d0 sp 00007ffff0cd99b8 error 4 in libcrypto.so.3[7fa6bb294000+26a000] likely on CPU 1 (core 1, socket 0)
Tue Aug 12 10:35:33 2025 kern.info kernel: [38671.912766] uhttpd[8486]: segfault at 7fea51f2ce60 ip 00007fea5285e4d0 sp 00007ffd632e1a58 error 4 in libcrypto.so.3[7fea52820000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 12 10:37:37 2025 kern.info kernel: [38796.186980] uhttpd[8606]: segfault at 7f034a6e53b0 ip 00007f034a91b4d0 sp 00007fff182f70b8 error 4 in libcrypto.so.3[7f034a8dd000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 10:37:37 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 119 seconds since last crash
Tue Aug 12 10:39:18 2025 daemon.err uhttpd[10667]: tr: write error: Broken pipe
Tue Aug 12 10:40:41 2025 kern.info kernel: [38979.862771] uhttpd[10667]: segfault at 7f8829c2dd40 ip 00007f8829e834d0 sp 00007ffcb61a92f8 error 4 in libcrypto.so.3[7f8829e45000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 10:41:24 2025 kern.info kernel: [39022.352602] uhttpd[15695]: segfault at 7f8362c86f80 ip 00007f8362ed24d0 sp 00007ffc9f31c3e8 error 4 in libcrypto.so.3[7f8362e94000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 10:41:46 2025 kern.info kernel: [39045.256895] uhttpd[16266]: segfault at 7f47b4eccef0 ip 00007f47b51d94d0 sp 00007fff1ac22188 error 4 in libcrypto.so.3[7f47b519b000+26a000] likely on CPU 1 (core 1, socket 0)
Tue Aug 12 10:44:36 2025 kern.info kernel: [39215.276213] uhttpd[16267]: segfault at 7f008877d9e0 ip 00007f0088a5d4d0 sp 00007ffdd2e271a8 error 4 in libcrypto.so.3[7f0088a1f000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 10:46:10 2025 daemon.err uhttpd[16648]: [info] luci: accepted login on / for root from 240e:b8f:5d1:be00:f319:199e:6492:56ad
Tue Aug 12 10:46:15 2025 kern.info kernel: [39313.716933] uhttpd[16648]: segfault at 7f097c790b90 ip 00007f097c9b64d0 sp 00007ffdc2c1ffc8 error 4 in libcrypto.so.3[7f097c978000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 12 10:46:29 2025 kern.info kernel: [39327.749687] uhttpd[17078]: segfault at 7fbd3e3d7dd0 ip 00007fbd3e5e94d0 sp 00007ffc19abb458 error 4 in libcrypto.so.3[7fbd3e5ab000+26a000] likely on CPU 3 (core 3, socket 0)
Tue Aug 12 10:46:29 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 9 seconds since last crash
[... cut ...]
Tue Aug 12 11:02:52 2025 kern.info kernel: [40310.750865] uhttpd[20893]: segfault at 7f6a44c43440 ip 00007f6a44f684d0 sp 00007fff51d1f008 error 4 in libcrypto.so.3[7f6a44f2a000+26a000] likely on CPU 2 (core 2, socket 0)
Tue Aug 12 11:02:52 2025 daemon.info procd: Instance uhttpd::instance1 s in a crash loop 6 crashes, 149 seconds since last crash

### ImmortalWrt version

r33247-467867283bb9

### ImmortalWrt release

24.10.2

### ImmortalWrt target/subtarget

x86/64

### Device

n2930

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

skefy, 2025-08-12:
兄弟可以参考下这个：https://github.com/immortalwrt/immortalwrt/issues/1429

1715173329, 2025-08-12:
Duplicate of #1429

closed 2025-08-12 by 1715173329

xcysy32, 2025-08-12:
> 兄弟可以参考下这个：[#1429](https://github.com/immortalwrt/immortalwrt/issues/1429)

看过，排除了证书问题，又发现我只要启停openclash就会炸掉uhttpd.最后写了个保活脚本