immortalwrt/immortalwrt #2008: 不能访问web管理页面

ammings, 2025-08-15:
### Describe the bug

pve中安装了snapshot版     immortalwrt-x86-64-generic-squashfs-combined-efi.img
shell界面可以进，ssh也能访问，apk update add等均可以
就是web管理页面访问不了

### ImmortalWrt version

snapshot

### ImmortalWrt release

snapshot

### ImmortalWrt target/subtarget

x86-64

### Device

pve

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2025-08-15:
snapshot 默认不带 webui。不知道 snapshot 是做什么的就别刷。

closed 2025-08-15 by 1715173329

ammings, 2025-08-16:
谢谢