immortalwrt/immortalwrt #2061: RK系列编译报错

VIKINGYFY, 2025-09-26:
### Describe the bug

日志节选：
```
2025-09-26T03:49:10.3640500Z rm -rf /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/boot /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/tmp/* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/lib/apk/db/*.post-install* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/usr/lib/opkg/info/*.postinst* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/usr/lib/opkg/lists/* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/var/lock/*.lock
2025-09-26T03:49:10.3658885Z Enabling usbmuxd
2025-09-26T03:49:10.3767709Z Enabling urngd
2025-09-26T03:49:10.3786964Z find /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-bfa51178/ -mindepth 1 -execdir touch -hcd "@1758737909" "{}" +
2025-09-26T03:49:10.3939450Z Enabling wpad
2025-09-26T03:49:10.4057324Z Enabling usbmuxd
2025-09-26T03:49:10.4248321Z Enabling wsdd2
2025-09-26T03:49:10.4339226Z Enabling wpad
2025-09-26T03:49:10.4393897Z rm -rf /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/boot /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/tmp/* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/lib/apk/db/*.post-install* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/usr/lib/opkg/info/*.postinst* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/usr/lib/opkg/lists/* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/var/lock/*.lock
2025-09-26T03:49:10.4526538Z find /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-02ebae08/ -mindepth 1 -execdir touch -hcd "@1758737909" "{}" +
2025-09-26T03:49:10.4629445Z Enabling wsdd2
2025-09-26T03:49:10.4767208Z rm -rf /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/boot /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/tmp/* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/lib/apk/db/*.post-install* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/usr/lib/opkg/info/*.postinst* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/usr/lib/opkg/lists/* /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/var/lock/*.lock
2025-09-26T03:49:10.4908026Z find /mnt/build_wrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/target-dir-7d206e0d/ -mindepth 1 -execdir touch -hcd "@1758737909" "{}" +
2025-09-26T03:49:10.8087823Z make[5]: Leaving directory '/mnt/build_wrt/target/linux/rockchip/image'
2025-09-26T03:49:10.8094469Z make[4]: *** [Makefile:23: install] Error 2
2025-09-26T03:49:10.8095056Z make[4]: Leaving directory '/mnt/build_wrt/target/linux/rockchip'
2025-09-26T03:49:10.8096959Z make[3]: *** [Makefile:12: install] Error 2
2025-09-26T03:49:10.8097541Z make[3]: Leaving directory '/mnt/build_wrt/target/linux'
2025-09-26T03:49:10.8100641Z time: target/linux/install#21.19#19.72#12.98
2025-09-26T03:49:10.8107985Z     ERROR: target/linux failed to build.
```

完整日志：
[ROCKCHIP _ immortalwrt_immortalwrt.zip](https://github.com/user-attachments/files/22561005/ROCKCHIP._.immortalwrt_immortalwrt.zip)

### ImmortalWrt version

r0-edea530

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

rockchip/armv8

### Device

NanoPi R5C

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

VIKINGYFY, 2025-09-27:
> 早上编译的一切正常
> 
> <img alt="Image" width="1032" height="933" src="https://private-user-images.githubusercontent.com/59593391/494661422-03bf8f91-a237-4bba-9597-4c740881c261.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg5NDMxMjUsIm5iZiI6MTc1ODk0MjgyNSwicGF0aCI6Ii81OTU5MzM5MS80OTQ2NjE0MjItMDNiZjhmOTEtYTIzNy00YmJhLTk1OTctNGM3NDA4ODFjMjYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI3VDAzMTM0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0ZmIwMzhiNThlMTQ5MmI4YzI5OWJlMjYzNjQ1ZDZjNjA4OGE5YzRhMzBiYWQzYjExZWM2ZTM4ZGU5OGY1MGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.kf-fxDEbxJRPMenlm55RF1w9wXSoyIOf7KrXMJZ2mq8">

你没添加wifi驱动吧

xlighting2017, 2025-09-27:
introduced in https://github.com/openwrt/openwrt/commit/04e9929c47e9fcf09bb32c44eb0a1447fd4962cf
and should be fixed by: https://github.com/openwrt/openwrt/pull/20211

as a workaround, you can add this to `.config`:
```
# CONFIG_WIFI_SCRIPTS_UCODE is not set
```

closed 2025-09-28 by 1715173329